ClockSelect/myevic #163: RX2/3: Overcharging batteries

phantasm, 2016-11-12:
I've managed to overcharge two HG2 batteries a little, so they were around 4.21-4.22 volts each, and after plugging them the device, it displayed "0% / check battery". I opened the BVO menu and, surprisingly, there were 3 battery meters while I only inserted 2 batteries. Readings were like "4.13v / 0v / 4.04v"

Maybe it's a hardware behavior, but maybe not. Seems that device relies on total voltage to detect 2 or 3 inserted batteries (like, if we have > 8.4 volts, it's 3 batteries) or something like that.

ClockSelect, 2016-11-12:
Strange.
Does the phenomenon cease when batteries discharge a little?

phantasm, 2016-11-12:
yep, after a short while

phantasm, 2016-11-12:
Now I've fully charged both batteries again, and got the same error. Playing with BVO didn't give any results.

phantasm, 2016-11-12:
Now, after some time the problem still exists. Flashing stock firmware didn't help. Maybe the device itself started to die? What can I do in this situation before returing it to the store?

phantasm, 2016-11-12:
Seems that the problem is related to back cover connectors, because it happens randomly when I attach/remove batteries and open/close the back cover.

Nabooh, 2016-11-13:
It's not related to the Custom Firmware, it's a well known hardware issue on the RX2/3, sadly...

Especially if you have 3 batteries showing in the BVO when you only have two inside the box, it means one of the battery connector wire of the Reuleaux has been broken...

Source for this issue ::
https://www.reddit.com/r/electronic_cigarette/comments/545m0s/rx23_check_battery_problem/
Pictures of this issue :
http://imgur.com/a/hjUKu

How to fix it:
https://www.reddit.com/r/electronic_cigarette/comments/545m0s/rx23_check_battery_problem/d80vl4j/
Pictures:
http://imgur.com/a/MwLXf

Well... Good luck !

phantasm, 2016-11-13:
@Nabooh thanks, it seems exactly like this :)

enricorov, 2016-11-13:
That sucks. Luckily, doesn't look too complicated to fix, if you have the tools at hand.

closed 2016-11-13 by phantasm