ClockSelect/myevic #165: RX2/3 Max watts?

Nabooh, 2016-11-13:
Hi guys!

So, it's not really an issue, more like a question about this firmware and the RX2/3.

ATM, I'm vaping on a dual coil at 0.513 ohms. With the dual 18650 back plate at 61W.

It seems I can't go over 70.6W, which is 6V in total. I know the original firmware can handle more than 70W in a dual batteries configuration.

Why can't I go over 70W?

Thanks!

bluenazgul, 2016-11-13:
6V is the maximum Output with 2 Batteries and 9V with 3 Batteries - this is the same on Stock Firmware and myevic

Nabooh, 2016-11-13:
Thanks, the more I thought about it, the more it seems logical to me. I didn't know it limits to 6 or 9V on the stock firmware too, but I was not used to vape at my actual Watts when I first used my Reuleaux ;D

This is such a great CFW, it makes the mod so convenient to use, getting in the menu is so clear and easy . Good job to Clock Select ;)

closed 2016-11-13 by Nabooh