ClockSelect/myevic #167: rx2/3 not able to use device monitor, just crashes

LockManipulator, 2016-11-14:
I have an rx2/3 and when I do device monitor, it has to be in HID mode for the program to even recognize it's connected, then it pops up a box saying I need VCOM mode as asks if I want to enable it. When I click yes, it switches my mod to COM mode and then goes back to not recognizing it because the program will only recognize it in HID mode. If I do device configuration, it needs to initially be in HID mode but then I can switch to COM mode and it will keep recognizing it. It won't initially recognize it if it's in COM mode though. Oh, and the whole program crashes at times when trying to do device configuration. Everything is up to date.

ClockSelect, 2016-11-15:
Several things, here.
HID is always on; it's what allows box commucication with something else. So whatever you choose as a setting, HID, COM, or whatever, HID is always on.
The monitoring works with COM. So for best compatibility, you may set the USB setting to COM. but anyway, COM may be forced by the monitor, so it shouldn't be an issue.
For things to work with the NFE (it may change in the near future), you have to enable NFE compatibility in the Expert menu: set "NFE" feature to "ON".
This feature is disabled by default because it causes issues with other softwares (like official updaters). As I said, we have projects to solve those issues, but it will take time. So by now, you have to ensure NFE is ON before connecting to NFirmwareEditor.
And of course, when you want to use another software to correctly communicate with your box, you have to revert NFE to OFF.

ClockSelect, 2016-11-15:
Oh btw: NFE crashes, sometimes.

LockManipulator, 2016-11-15:
The NFE is ON and has been but it still doesn't recognize the mod is connected when I have COM on. Only HID so HID is not working when COM is switched on. And NFirmwareEditor always crashes when I click device monitor and then cancel. When I have the mod with HID mode and I click device monitor, then it will see it and ask if I want to switch to VCOM mode, if I click yes, it changes to COM (even though it will still say HID on my mod, but if I click fire it changes to DSK which is after COM) and then will say no device recognized because NFirmwareEditor is not detecting the mod in COM mode for some reason. The fact that the mod still says HID after NFirmwareEditor changes it is strange too, because clicking fire will change the mode to DSK. Normally it's HID --> COM --> DSK.

ClockSelect, 2016-11-15:
Leave it on COM, it should work.

LockManipulator, 2016-11-15:
It's actually not being recognized at all on COM. As if nothing is plugged in.

ClockSelect, 2016-11-15:
One thing I learned with NFE is that each time to touch something, better close and relaunch it.

LockManipulator, 2016-11-15:
Same issue. It just doesn't see the device when it's in COM mode. Restarted my computer and the mod, still doesn't change anything.

ClockSelect, 2016-11-15:
Maybe you have an issue with the COM driver.
Did you install a driver for the USB COM to work?
See in the driver/ directory for more information.

LockManipulator, 2016-11-15:
That's strange, there's no usbdriver.sys in my system32/drivers. I'm trying to google how to download it but can't find a definitive answer.

ctrlaltca, 2016-11-15:
The correct name is `usbser.sys`.
I'm experiencing the same problems with my RX75. Config load works, config update fails, device monitor crashes.

ClockSelect, 2016-11-15:
Maybe you should try a [nightly build](https://ci.appveyor.com/project/TBXin/NFirmwareEditor/branch/master/artifacts) of the NFE.

ctrlaltca, 2016-11-15:
Bingo, device monitor works using a nightly build.
Config upload still fails, here's the log:
```
Set Param command - Start Addr: 0    Param Len: 2048
Set Sys Param complete.
	Company ID ............................ [0x000000da]
	Device ID ............................. [0x00002980]
	Product ID ............................ [0x00845201]
	u8UpdateAPRom ......................... [0x00000000]
Incompatible parameters format.
```

ClockSelect, 2016-11-15:
@ctrlaltca : Yes, compatibility with the NFE is only partial.
You may use the Device Monitor, Battery profile editor, COM Terminal, Screenshot function, but not (yet) the configurator.

LockManipulator, 2016-11-15:
Ah dang I can't disable driver signature since my laptop screen is broken and during that screen the laptop can't display on another screen. I tried to use command line for it but it didn't do anything. I tried the nightly build, same issue, it just doesn't crash after clicking cancel. But it's still not seeing anything plugged in when my mod is on COM mode. It might not be possible for this to work for me though if I can't disable driver signature.

ClockSelect, 2016-11-16:
-> @TBXin

closed 2016-11-16 by ClockSelect