ClockSelect/myevic #168: setting up cygwin to compile source in windows

RenegadeTitan, 2016-11-15:
alright, so here we go. hopefully this thread helps others who may want to dive in head first and help contribute to this epic project. quick shout out to all the giants whose shoulders i hope to stand on that made all this progress possible. absolutely amazing work.

my goal is to be able to edit firmware source code then compile it into a working firmware.bin file. i am starting with windows 7 and a basic hex editor. i am going by the instructions listed in ReservedField's evic-sdk github page in his readme file.

i have made it thru the instructions but i think i missed something in the first step (and dont yet know what else i may have missed). it says "install the precompiled arm toolchain". it says use a path without spaces to avoid problems with the build process. i ran the precompiled installer. i think i might have to point cygwin to this directory somehow?

i followed all other steps and received no errors until i changed directory to evic-sdk and typed "make". i recieved 2 errors:

XXX@XXX-PC ~
$ cd evic-sdk

XXX@XXX-PC ~/evic-sdk
$ make
/bin/sh: arm-none-eabi-gcc: command not found
/home/XXX/evic-sdk/make/Common.mk:91: *** Could not detect arm-none-eabi library paths.  Stop.

ClockSelect, 2016-11-15:
You don't need the SDK for the firmware.
So if the SDK does not build, it's not an issue.

RenegadeTitan, 2016-11-15:
ok. what exactly do i do to compile your source the easiest way)? same procedure as i used to prepare evic-sdk?

ClockSelect, 2016-11-15:
You have to define the ARMGCC environment variable.
Add:
export ARMGCC="/cygdrive/c/GNUARMEmbedded/5.3.2016q1"
(or whatever the install path is)
to the .bashrc file

RenegadeTitan, 2016-11-15:
do i do that from cygwin with an echo command or manually into .bashrc like thru notebook? if manually, at the end? do i need a # symbol first? (it looks like all other lines start with a # symbol)

ClockSelect, 2016-11-15:
add it anywhere in the .bashrc (end is fine) without any # 
yes, /cygwin/c stands for windows C:
adapt your path as needed.

RenegadeTitan, 2016-11-15:
so that seems to correct path for arm-none-eabi but still getting error for arm-none-eabi-gcc:

cHAOs@cHAOs-PC ~
$ cd myevic

cHAOs@cHAOs-PC ~/myevic
$ make
make: arm-none-eabi-gcc: Command not found
arm-none-eabi-gcc -Wall -mcpu=cortex-m4 -mfpu=fpv4-sp-d16 -mthumb -Os -fdata-sections -ffunction-sections -std=c99 -fno-builtin-printf  -IC:\cygwin\home\cHAOs\evic-sdk/nuvoton-sdk/Library/CMSIS/Include -IC:\cygwin\home\cHAOs\evic-sdk/nuvoton-sdk/Library/Device/Nuvoton/M451Series/Include -IC:\cygwin\home\cHAOs\evic-sdk/nuvoton-sdk/Library/StdDriver/inc -Iinc -Isrc -IC:\vape\5.4.2016q3/arm-none-eabi/include -c C:\cygwin\home\cHAOs\evic-sdk\nuvoton-sdk\Library\Device\Nuvoton\M451Series\Source\system_M451Series.c -o C:\cygwin\home\cHAOs\evic-sdk\nuvoton-sdk\Library\Device\Nuvoton\M451Series\Source\system_M451Series.o
make: arm-none-eabi-gcc: Command not found
make: *** [Makefile:183: C:\cygwin\home\cHAOs\evic-sdk\nuvoton-sdk\Library\Device\Nuvoton\M451Series\Source\system_M451Series.o] Error 127

also there is no system_M451Series**.o** in that location, only system_M451Series**.c**

ClockSelect, 2016-11-15:
make: arm-none-eabi-gcc: Command not found

means it's not the good path.

RenegadeTitan, 2016-11-15:
eurika! arm website said specifically not to check box to add environmental variables but i tried it anyway. and i re-installed it directly into the default cygwin/home/user directory so i could remove that part we added to the .bashrc file. compiled without errors. thank you for your help.

ClockSelect, 2016-11-16:
I'll keep this issue opened some time since... you know... people may want to have time printed into the glass of their tank.

closed 2016-11-16 by RenegadeTitan