ClockSelect/myevic #172: Suggestion: Preset resistance in VW mode

Kulimeya, 2016-11-15:
It would be great if I could preset the resistance in the coils section for wattage mode. I have (common) problems with my 510 port on my Cuboid, where the resistance often gets detected wrong. 
I already burnt some coils, because it got detected as .9 instead of .4, but usually it gets detected as .5, .6, .. . I really hoped your firmware includes this feature already, I hope you will add it. 

Thanks!

**Box Model:**
Cuboid

**Coil setup:**
Cleito .4

ClockSelect, 2016-11-15:
I'd not swear it'd be the safest option.

afol2, 2016-11-15:
i have found that the cuboid Vs vtc mini, the cuboid tends to be more susceptible to any change in the coil, they never match the Ohm, and if there is a little bit of for example a lead that have a hotspot, i goes right away to atomizer short.

Kulimeya, 2016-11-15:
Maybe in the expert section? It would be great because it would fix my problem. :)

afol2, 2016-11-15:
i know it could be used for bad things, for example, if i had a 0.09 ohm coil, i could preset it as a 0.25 ohm, it will fire, but the amp limit will be bypass and the lecture its not going to be accurate, maybe damaging the Mod.

SEEKNDESTROY, 2016-11-15:
@afol2 Or worse, venting batteries.

Kulimeya, 2016-11-15:
Oh I thought the amp limit is a hardware limitation.
What about a voltage mode? There's an Aspire mod which has that.

afol2, 2016-11-15:
@Kulimeya have you try to fix the 510 port with the common fix for the cuboid ? 

[https://www.youtube.com/watch?v=jQ-gs6EQN3U](url)

Let me tell you, i did this to a kanger tank that i used to have, and it really solve the problem

afol2, 2016-11-15:
@Kulimeya yes its a hardware limitation, but blocked by the software.. to avoid the damage of the hardware.

afol2, 2016-11-15:
@Kulimeya about the VV mode, i was going to ask that same question to @ClockSelect, back in the good ol'days, we used to have VV or VW mode, in the "EVIC", I think it could open more possibilities

bluenazgul, 2016-11-15:
Why would you need VV when you have VW? When you change the W you will also change V

afol2, 2016-11-15:
If you have a Mod with problems with the lecture of the coil, if you select VW, the voltage will go up and down depending of the resistance, instead if you have a VV, your voltage will be the same no matter what lecture of the resistance it gives you, he can read whatever he wants, but the Voltage will be the same so in theory, it should gives you a more stable vaping **Just in the case your mod its not reading accurately your ohms** @bluenazgul

afol2, 2016-11-15:
Think of it as a "Locked" By Pass mode @bluenazgul, by locked i mean, you put the voltage... and it will be the same always until low batt of course hehe.

ClockSelect, 2016-11-15:
@afol2 : if you keep voltage constant over a fluctuating resistance, what will the power do? will it be steady stable?

If the box reads .9Ω instead of .4Ω, it's not because of a *bad reading*. It's because the whole circuit made of the 510 connector, atomizer and coil *actualy has* a .9Ω resistance. So *something* is adding .5Ω to the circuit. If you send power in there, the .5Ω *something* will dissipate more power than the coil.

And I'm more sure you want *something*, *somewhere* in your box dissipating more power than your coil. Or do you?

afol2, 2016-11-15:
@ClockSelect True.. haven't seen it that way..

afol2, 2016-11-15:
but i dont know why, with the evic no problem at all, cuboid, some times it gives me that problem (not that big of a difference like  .5Ω, but some times just .05 or .1Ω) using same atomizer, same build

Kulimeya, 2016-11-15:
I found that video, too. That fix worked surprisingly well, but after some weeks it stopped working. No atomizer found if I add the ring.

ClockSelect, 2016-11-16:
So... do we agree the suggested software solution is not a solution?

closed 2016-11-17 by ClockSelect

forkoz, 2016-11-26:
lol I see. its because idiots.

My case is coil starts at 1.8 and degrades to 2.5. I want to keep firing between 2.8 and 3.3V at one setting to pace the battery, guess I'll have to code it myself. Quite funny that the reverse of what you warn of happens to me; VW pumps up the power and I want less.