ClockSelect/myevic #175: function of click menu: clock

RenegadeTitan, 2016-11-15:
i think if when, for example, setting interface-clicks-2-clock (default) is set, 2 clicks should switch between clock and info on main screen (as it does) but also switch between 3D and info on main screen (when screen-misc-3D is set).

so in other words, under clicks 2 3 4 menu, clock setting should switch from clock to info and also from misc-3D to info (and back).

in other other words i like having the 3D objects covering half my mainscreen but i would like to click twice to then see my mainscreen and click twice to cover half mainscreen with 3D objects again. just like i can do with clock.

its a small thing but i think it would benefit all who like the 3D objects on their mainscreen or would like them but dont want to block out info when not firing.

ClockSelect, 2016-11-16:
please...

closed 2016-11-16 by ClockSelect

RenegadeTitan, 2016-11-16:
lol. too small of an issue? i thought it might be a quick copy paste of code to do with your next release. all good.

ClockSelect, 2016-11-16:
Vape. If tomorrow you tell me that you spent all your kho-gen-do to unexpected 150W flame burst while testing a comfortable vaping experience, I'll tell you you don't need clock or 3D any more. And, I'll don't have to tell you. you'll know.  You do not (yet) imagine the quantity of cotton the world owe me to have a clock.

RenegadeTitan, 2016-11-16:
on behalf of all the users of myevic, thank you for all the work youve done with this project.

ClockSelect, 2016-11-16:
Hey! didn't wan't to be rude. You closed everything, just like you're leaving.

RenegadeTitan, 2016-11-16:
maybe i misunderstood. seemed like your last 3 posts in different areas indicated i was annoying you in those threads. i will continue to open threads if you find it helpful (but they will most likely be smaller things, the firmware is looking good). or maybe a thread asking questions about the source code (if you arent too worried ill blow my face off lol) feel free to open up any thread i close if you think it is still productive.

ClockSelect, 2016-11-16:
Well sure I can't learn you C, so I cannot guanrantee you won't set your box on fire. But you know this already.
PID was an achievement, seems that one people out of 5,000 found it usefull. Thanks you've been there, got one user^^
Hm, advice if you wish to upload something you've done to your box: have two boxes^^

RenegadeTitan, 2016-11-16:
lol. yeah, were on the same page.

see you in the next thread