ClockSelect/myevic #176: wismec presa75tc problem reading ohms

lella88, 2016-11-16:
Hi, I wanted to point out that in wismec ohms are rievati abnormally.
On the main screen they are detected in .223 ohms instead of 0:22 ohms.
thank you so much

ClockSelect, 2016-11-16:
Do you mean it reads ohms better than before?

MaRCoMLXXV, 2016-11-16:
There's nothing wrong with the resistance display and/or measurements, @lella88. It's just displayed more accurately, showing three decimals (thousandths of an Ohm) instead of the two decimals (hundredths of an Ohm) that you're used to see in the stock firmware. The higher accuracy is by @ClockSelect's design.

lella88, 2016-11-16:
Thanks a lot for the answer.
It was strange to see the reading ohms that way.

ClockSelect, 2016-11-18:
It's been made for low TCR metal like SS so you have a reasonable accuracy on vape temperature.

vaperjack, 2016-11-18:
Hab das gleiche problem bei der RX200S da ist vorher 0,25 Ohm gestanden und jetzt steht .251 ohm

bluenazgul, 2016-11-18:
@vaperjack "it's not a Bug, it's a feature" 

Die Anzeige wurde optimiert um 3 Stellen nach dem Komma anzeigen zu können,  dieses bringt gleichzeitig eine Verbesserung im TC Bereich mitsich, wird sogar in der README.md und README_de.md beschrieben.

closed 2016-11-21 by ClockSelect