ClockSelect/myevic #177: Suggestion: Custom Amp Limit (Battery Profiles)

bluenazgul, 2016-11-16:
Custom Amp Limit (for specific Batteries)

We all know the Cuboid has an 25 Amp Limit (by Software) but what if you use Batteries like Sony VTC4/VTC5/VTC6 (30A max) or Sony VTC5a (35A max) would it be possible to raise the limit then to fit your Batteries?

**Box Model:**
Cuboid (maybe other Devices too)

**Coil setup:**
any

PS: for Users own risk "a hidden" "Custom Amp limit (CAL)" Setting (pulsed Amps are most more)

StepWi, 2016-11-16:
There are other components inside the mod that limit the max amperage, e.g. the wires, the battery connectors, the pcb, the MOSFet, ..., so you risc a hardware defect (or something worse) if you exeed the manufactorers limitations.
I even don't like the max settings of the protection time to 15 seconds, as nobody knows, if there are internal parts generating heat or suffering if used 15 secs under high load instead of 10 secs.

RenegadeTitan, 2016-11-16:
also i would like to add that the vtc4 and vtc5 are mistakenly quoted as having higher than 23a max continuous discharge ratings because higher amps are used in testing data sheets from the manufacturer. but if you look at operating temperatures at various amperages the max contimuous discharge for both is actually 20-23a. and 23a is the highest that ive seen from any 18650 tested. all will do short bursts at crazy amps but will incur shorter lifespans. i dont know specifics but there may also be dangers if operating an 18650 above its max continuous discharge ratings more than a second or so due to higher operating temperatures. i think 25a is a safe limit until someone invents a more powerful 18650.

afol2, 2016-11-16:
Right now there is only 2 batteries capable of get a continuous 30 amps without venting, LG HB4 or HB6, 30amps @ 1500 mah, tested by the Redditor, Mooch315. VTC 4 only 23 amps and VTC5**A** 25 amps, saying that, i open an issue before, for this exact thing, wanted the cuboid to handle a little more, but, we dont know how capable is the device internal to support that amp. wish we think its almost at its limits. but yes, it is doable by software, but its the hardware that I dont think it can resist @bluenazgul, remember all current goes trough the Shunt Resistance, Cuboid only have 1, i think the RX have 2.

bluenazgul, 2016-11-17:
I had the old "51A Patch" a long time on my Cuboid - never had any issues (used about 35 Amps) Coubid still works (only has a broken Contact to check volatage between the two batteries)

afol2, 2016-11-17:
@bluenazgul that info, its helpful. at what ohms do you used ? and what wattage ?

bluenazgul, 2016-11-17:
RTA at 0.16 Ohms was the lowest i used at maxmium 200W - also Cthulhu RTA III at 0.23 Ohm with 200W

Maybe only an Expert Option to set Amp Limit to a Value with Maximum 35Amps

Could help a lot TFV User

afol2, 2016-11-17:
@bluenazgul last time, i told @ClockSelect if maybe, only selecting some kind of trusted battery it can be enable, for example, LG HG2, max 25 amps (like normally will do), LG HB6 or HB4 it can unleash the amp limit to some trusted point..

ClockSelect, 2016-11-17:
Custom batteries actually have Amp limits; just the battery profile editor does not let you edit it yet. So the xutom battery profile has a 25A limit for now, until the editing field is added to the editor.

cadmaro, 2016-12-04:
@bluenazgul
Where can I find this 51a patch?

bluenazgul, 2016-12-04:
Nowhere, it was an exidental release a long time ago and was removed realy fast. Because it was an real security Problem

cadmaro, 2016-12-04:
Well that's a shame. I really like my cuboid and this firmware is awesome but I have a few coils that could use a little extra power

closed 2016-12-21 by ClockSelect