ClockSelect/myevic #179: Function "Freezing" before changing the batteries on the mod without RTC.

Villista, 2016-11-17:
**Box Model: RX 2/3**

**Coil setup: A few different!**

Waping enjoyable for all! Probably many people, like myself, use extra charger and during the day replace the batteries in their boxes. While in the boxes, where there is no RTC, reset the time and counter of liquid! Time can be set again, even though it is exhausting. But the counter of liquid is lost forever! I propose to add a function to "Freeze" in the menu (in the menu "Experts" for example)! This function records information about time/date and the counter ml of liquid in any place of any available memory and turn off the device. We quickly change the battery and when starting device checks if data is stored, and then restores time/date and counter of liquid, then deletes the data and continues to load in standard mode! Perhaps in addition to these so you can save some data that is a pity to lose when changing battery. A couple of seconds for the backlog hours at each change the battery I think is irrelevant. What do you think about this?

drminol, 2016-11-17:
Nice think! I'm waiting for implement it.

ClockSelect, 2016-11-18:
Right solution would be to store liquid among with puff counters, but it's a bit more tricky than it sounds. It's on a todo list somewhere...

Villista, 2016-11-18:
I would be very glad to implement this supplement. :-)
I change the batteries 2 - 3 times a day (my batteries 1500 mA) and each time set the clock and date is very problematic! And the counter liquid are losing too!