ClockSelect/myevic #180: Adjusted Watt vary when vaping...

badummtss, 2016-11-17:
Hello ClockSelect, 
i have a Problem since yesterday, after i update on the newest version (161115), my adjusted watt settings in powermode vary when i vape.

bluenazgul, 2016-11-17:
@ClockSelect here are the Videos from @badummtss Issue (the Videos are created by him)

The Video with the Issues
[Video with issue.zip](https://github.com/ClockSelect/myevic/files/597098/Video.with.issue.zip)

On older Firmware (befor the Power Curve Feature) without this issue
[Video without issue.zip](https://github.com/ClockSelect/myevic/files/597104/Video.without.issue.zip)

acevalsl, 2016-11-17:
same error with etc two mini

acevalsl, 2016-11-17:
video with jumping wattage
[IMG_2631.MOV.zip](https://github.com/ClockSelect/myevic/files/597130/IMG_2631.MOV.zip)

fredsen1337, 2016-11-17:
Same problem with rx200s. Seems the curve funktion is not disabled properly. Before you added the feature there was no problem with the wattage.

filippo90, 2016-11-17:
Here on RX 2/3 same issue of video!

StepWi, 2016-11-17:
Just checked on Evic Basic with 2.6 Ohms at 10.2W: resistance and watt is oszillating a bit, roundabout 0.1 up to 0.2W, and less than 0.1 Ohms.

afol2, 2016-11-17:
Same problem here with Cuboid and Evic VTC Mini, wasn't realizing that before i see this post.

falconettt, 2016-11-17:
Don't bash me, boys, -)
but nothing bad happened to the curve:
[image]

Yes, the power curve seems has some noise oscillations in first decimal.
But I like it! I'd be more glad even more controlled oscillations!
It's not a bug, it's feature (IMO)

afol2, 2016-11-17:
> It's not a bug, it's feature 

Its like now is more accurate ? now you know the true power it is delivering ?

StepWi, 2016-11-17:
At me it doesn`t look more accurate. Resitance is counting very fast from 2.58 to 2.61 and backwards. As we are talking at about 1 or 2% variation, it does not affect vape temperature - therefore I don't see this behaviour as feature, but as bug. But I have noticed this behaviour only because of readig this thread.

ClockSelect, 2016-11-17:
It's just that know the actual wattage is displayed instead of the setting wattage - that we don't really care about. So there's some noise, inevitably.

badummtss, 2016-11-18:
@StepWi you really mean that there is no worry about that? 
And @ClockSelect what i read out is, that this is more a feature than a bug?

StepWi, 2016-11-18:
With the explanation from ClockSelect, that the actual wattage is displayed (feature), and the derivation of less than 2% I don't see any problem.
Remember, that the measurement of the resistance probably is worse than 2 %, you have internal wires, the 510 connector, wires and screws in your Atomizer, ..., so there is nothing to worry about in my opinion.

ClockSelect, 2016-11-18:
I'd says that even if it was ±5% it shoudln't be too much of a concern.
Some boxes are good at reading numbers, others are bad, it depends on model and setup.
Anyway, showing setting numbers instead or real numbers does not solve the problem, it just hides it. But I guess it's less scary^^

badummtss, 2016-11-18:
Thank you Guys for your explaination about this. 
Ok then for me i have a disco lighter instead of a Mod Box ^^.

Great Job but before i eat fish i would drop the fishbones away - you know what i mean ;)

Thank you Very much for that great piece of software @ClockSelect .

closed 2016-11-18 by badummtss