ClockSelect/myevic #181: RX2/3 2-3 cell box freezed

neodjoo, 2016-11-17:
Hey all, first of all, thank u all for the contribution, i ve set the myevic firmware in a RX2/3 box, it work fine in 2 cells than i used the 3 cell after that i went to the 2 cells the box wont work, it seems like the box can't read the 2 batteries, can't vape nothing, any issues?thanks a lot
[image]

ClockSelect, 2016-11-17:
A similar issue has been discussed here: #163
I don't know if ti will help.

Filo301, 2016-11-18:
I had this same issue. Top balancer cable was broken.

neodjoo, 2016-11-18:
ok thanks for the contribution, seems like it's the same case!

closed 2016-11-21 by ClockSelect