ClockSelect/myevic #182: Removing Profile?

tharealgc, 2016-11-17:
Hello,

FANTASTIC, AMAZING and MIND-BLOWING Firmware dude, it did everything I wished for original software to do!

Only thing I am missing is removing a profile the rest is just perfect!

One more thing before I go, I only changed 1 setting in expert which was putting my battery info in which are VTC5 and my battery life is halved now (MOD: RX2/3), don't know if its related yet, will keep testing!

And again Thank You very very very much for your hard work and making all of us a very happy vapers!

ClockSelect, 2016-11-18:
For now you can't remove profiles, but you may use the multi-click feature to cycle through profiles without saving the current profile, so you don't overwrite coil values (that the difference between cycling with multi-clicks and selecting via menu).
Selecting a battery model changes the way battery charge % is computed, but not the actual charge of the battery. The number shown is just more accurate.

tharealgc, 2016-11-18:
Ahhha, so battery setting it's basically only for calculating remaining battery level and not to optimize the discharge...
In any way I am really happy and thankful for your work with this FW, it's amazing and the sweet Algo is SWEET!
Thank you.

tharealgc, 2016-11-19:
Hello again,

After further testing I can say I really think this FW drains a bit more power... Maybe it can be the Sweet Algo I have chosen?

On both BAT-GEN and BAT-VT5 it drains the battery to the same level as stock FW, but from stock FW I had around 200 seconds more (45 Puffs).

For testing I have used same coils and same TC settings and the results are:
Stock FW: Approx 200 Puffs and Runtime is around 850 seconds.
MyEvic FW: Approx 150 Puffs and Runtime is around 670 seconds. (Same results with both BAT- Presets GEN and VT5)

What could cause this?
I will keep testing and see if I can pinpoint what is causing this, but if any of you guys might give me a hint, please do so.

bluenazgul, 2016-11-19:
Cant confirm it by myself, but friends say Cuboid and RX200S also Do have a much higher Battery drain since the flashed myevic, i personaly cant confirm a higher drain on VTC mini with only one Battery

RenegadeTitan, 2016-11-20:
@tharealgc same setup (rx23 - 2 batt config) i would think i would notice a 25% decrease in capacity although i do change settings/coil size/etc quite often so may not have noticed. please keep testing becuase if there is a significant difference it would be helpful to know that.

i believe its the algo-sweet. this algo seems to perform quite different than stock firmware. maybe try testing with algo-off (this algo is similar to stock firmware) to see.

one thought was screensaver or screen dim time. should be negligible but do you have screensaver set to 30 minutes or anything like that? how much time does it take to go through 150 puffs? just trying to rule out differences in use from old firmware to this one.

if your testing seems to confirm increased drain i can test it too. but i would prefer to never load the old firmware again lol

tharealgc, 2016-11-20:
Still got around 20% less with Algo Off...

RenegadeTitan, 2016-11-20:
ok ill test it too. TC SS 316 also. batteries last me a little over a day so ill have results in 2-3 days or so.

never used puff counters or timers before. pretty cool feature. love how you can zero it by selecting it and long pressing fire. great functionality.

RenegadeTitan, 2016-11-21:
ok so i recently switched to much higher nic content to cut down on pg vg and my batteries are lasting many days between charges now so test will take a very long time for me.

one thought i had was when your ending your test. is it when battery symbol starts flashing or box wont fire any more or at a specific voltage?

tharealgc, 2016-11-21:
I drain my batteries to 5%, when I put em in charger they rate at 3.37v, I had the same voltage when I drained them on stock FW to 10%... I don't use any correction just the preset in Expert there is a battery profile VT5 for my VTC5 batts... so to clear this I put my bats into mod fresh when they are 4.20v and drain them to 3.37v, with stock FW I get around 40-50 puffs more which is around 150-200 working seconds more than with myevic firmware...

I don't want to switch back to stock fw by any means, but I still hope we can get to the source of what is draining our batteries quicker. If we get the same runtime of bats with this FW it will be UNBEATABLE!

RenegadeTitan, 2016-11-21:
ok thanks for quick reply with all the necessary info. that rules out that idea. what about base ohm reading? are you using the more accurate 3rd decimal in new firmware?

tharealgc, 2016-11-21:
Yes I do, but I also use the same coil which is 0.16 and in this FW is used as 0.163 coil...
I don't really think that 0.003 ohm difference can drain 20% more battery...
All the rest of preset is the same: No Preheat, 75W, TC - SS 316...

RenegadeTitan, 2016-11-21:
i think that may be it actually. with SS these differences are exaggerated. that .003 translates to about ~30 degrees+ hotter when firing at same temperature setting (estimate, not exact figure, i will replace with exact figure if i can figure out the math lol). maybe more. thats about 10% more of a temperature change which could easily translate to a 20% increase in power consumption. the old firmware fired way below set temp due to rounding down base ohms. you can set base ohm manually to previously set reading or what i recommend is lowering temperature around 30-40 degrees untill you sense a comparable vape to previous firmware. or enjoy the warmer denser vape using more power lol. please feel free to test this to confirm.

tharealgc, 2016-11-21:
I don't say I am measuring the temp good while I vape, but I got it to the exact same temp I had before on old FW, also I adjust TCR value for the TC to match the experience and temp I experienced while vaping on stock, so for me the vaping experience on both FWs is kind of identical.

I can set my resistance to exact 0.16 and set TCR to the same as it was on stock fw, will probably attempt that when I swap to fresh bats next time.

My other question is algorithm... when its off it should use the one of stock firmware, but is it? I mean is it exactly the same as stock RX 2/3 one? or does it mean that it's the stock one from eVic? This also might be the case...

For more testing purposes we could just vape in Power mode say at 50-60w on both fws and see if the results shows the same puff/burn durations...

RenegadeTitan, 2016-11-21:
i can test output voltage from device in VW mode. still need same base ohm readings as device regulates wattage by adjusting voltage to match ohms.

tharealgc, 2016-11-21:
I didn't mean to measure actual output in VW... What I meant is take a coil and vape on it from 100% battery to 5% battery and compare how many puffs or even better burn seconds you get from both FWs...

RenegadeTitan, 2016-11-21:
what were talking about here is either  a lot more power used during short use or a little power lost during long sleep. any efficiency issues would be noticed as internal temperature increase. so power is either going to coil during use due to differences in settings or slowly dissipated during sleep. over night (8-10h i dont lose even .01 volts so its not that. power must be going to coil. higher firing temp or slower warmup are biggest power consumers. higher base ohms, higher temp setting, lower set wattage in TC would all explain this. im sure its just differences in settings. if somehow you find that with all the same exact settings power consumption is different, we can try to find cause but i dont see how it could.

tharealgc, 2016-11-22:
Same goes for me, I am puzzled of how I get 20% less battery life with all the same settings...
Maybe it is that stock FW was not so accurate? but then the accuracy goes from the chip which is still same so it should be same... I hope that I don't start pulling my hair out soon...

I think we test it a bit more and if we don't find any clues we should just give up on figuring it out or it can become maddening...

RenegadeTitan, 2016-11-22:
i have a habbit or editing most my posts. please see what i added above. i believe that last paragraph is your solution.

tharealgc, 2016-11-22:
Well your edit is just about right, I use 117 TCR and will try lowering my coil to exact 16 as it was on stock fw for testing )))

ClockSelect, 2016-12-29:
@tharealgc what did your tests showed?

About profiles: a profile editor should be available soon, stay tuned.

tharealgc, 2016-12-29:
@ClockSelect nothing much actually I tested all Algorithms and the drain is the same, I don't know if reverting to stock would change something, I would like to test it but it's a shame to go back to stock after this fantastic firmware... also I changed my coils recently so testings won't be accurate anymore. It's all good I guess, and it's great that you will include profile editor.

Sorry I could not provide any data, but it's the same as it was when I first posted first test results.

closed 2017-08-30 by ClockSelect