ClockSelect/myevic #184: Suggestion: At power settings over 100 watts, drop the ".0" and keep the same font size

Justin-L-Franks, 2016-11-19:
I just installed this custom firmware on my Cuboid and was immediately impressed! Little things I found annoying (like the scroll speed for power settings, and the scrolling continuing in 0.1 watt increments even when making large changes) are fixed, and the extra features are quite useful. Thank you so much for your work on this firmware.

I have a suggestion regarding the wattage displayed in power mode. Under 100 watts, power is adjustable in 0.1 watt increments, but over 100 watts, this changes to 1 watt increments. This is good. But with the inability to set a power over 100 watts with tenths of a watt means that the ".0" after the displayed power is unneeded. And because of the width of the display, a smaller font must be used to fit four characters.

This should be changed so that the power displayed over 100 watts should only use 3 characters, and ditch the superfluous decimal and tenths of a watt. By doing this, only three characters will be needed to display any power setting, and the font size will be consistent throughout the entire power range.

Thanks for your consideration regarding this minor annoyance.

**Box Model:** Cuboid

SEEKNDESTROY, 2016-11-19:
Try: Menu> Interface> 1C5F> On

> Enable increment of temperature setting by ±1°C and ±5°F instead of ±5°C and ±10°F

Justin-L-Franks, 2016-11-19:
Once you get to 100 watts, the increment is ±1 watt regardless of the interface setting. But I tried turning that setting on anyways, and the behavior is still the same -- the ".0" is always displayed, which means that if you set it to 100+ watts, the font size has to be decreased to fit the fourth character.

It doesn't make much sense to reduce the font size and display the ".0" above 100 watts when you can never set anything other than an integer wattage regardless of the "1 Watt On" interface setting.

ClockSelect, 2016-12-29:
I tried, and I find the result a bit inaestethic and confusing.

[image] [image] [image]

It becomes unclear at one glance if we're at 15.0w or 150w.

Justin-L-Franks, 2016-12-30:
It seems clear enough to me. Would adding a menu option to choose either behavior be possible? If not, and if you are not going to release this tweak, would you be willing to share the changes so I could compile it for myself? I haven't taken more than a quick glance at the code, but from what I saw, this would be done with a modification to the Value2Str() function in display.c. I could then just add those changes in for myself each time there's a new release. Even if future commits make changes to that function I'll be able to figure out how to get this in myself. It's been a while since I've done any C++ programming, but that shouldn't be an issue.

Thanks for taking the time to look into this suggestion.

ClockSelect, 2016-12-31:
in mainview.c lines 505/506, swap the commented lines.

ClockSelect, 2017-01-12:
Done

closed 2017-01-12 by ClockSelect

Justin-L-Franks, 2017-01-16:
Thanks so much! I installed the latest binary and was pleased to see that this tweak was added. I can confirm that this works perfectly. Having the same font size throughout the entire power range is much more consistent, and at >= 100 watts the selected power setting is easier to read due to the increased font size.

Best regards,
-Justin L. Franks