ClockSelect/myevic #186: Games On Evic Vtc Mini

anderlobo, 2016-11-20:
https://www.youtube.com/watch?v=1wyj3uMJNE8 Google browser game, no internet connection.
https://www.youtube.com/watch?v=8Q2mwnvBaGQ Snake on e-cigarette (eVic-VTC Mini).

It would be possible to do one of these, I imagine it should weigh a few kb.
thanks...

drminol, 2016-11-21:
Do you have a smartphone? Who the hell are playing games on the ecig?! It hasn't any sense...

marvinerzz, 2016-11-21:
I have a smartphone, but it'd be nice to have a few games on my vape every now and then for lulz

LockManipulator, 2016-11-21:
I agree, I love having flappy bird and would love more games too.

ClockSelect, 2016-11-22:
I'm afraid I won't use the last few available kB for that.

Daveee10, 2016-11-22:
@ClockSelect: The data storage area can't usable for store the settings? Maybe something movable from dataflash.

closed 2017-01-18 by ClockSelect