ClockSelect/myevic #187: Wismec Reuleaux RX2/3 not waking up on USB connect.

ajape1, 2016-11-21:
Hey all! My Reuleaux RX2/3 is NOT waking from sleep mode when you plug the USB in, is this happening to anyone else? Because with MyEvic firmware I always need to wake it up manually by clicking the fire button once after I plug it to the USB, otherwise it doesn't connect to the PC or start charging and the screen stays off.  With the Wismec firmware, My Presa 75TC, my RX200 and my RX2/3 all wake up from sleep immediately and start charging as soon as I connect them to USB either from the PC or to the power adapter, and the screen then turns off after 30 seconds just leaving the filling battery animation until the battery is full, then the screen turns off completely again. Is anyone else's experience with the Wismec or the MyEvic firmware any different? Thanks, guys, and most especially thank you ClockSelect team for taking the time and making all these efforts on all our behalf, they are enormously appreciated!

**Box Model:** Reuleaux RX2/3

**Coil setup:** 2x32AWG SS316L 3mm coils @0.3omhs each on the Avocado 24.

SEEKNDESTROY, 2016-11-21:
I cannot confirm this behavior with my RX200S. My RX200S starts charging as soon as I plug it in when UCH is on. The only time my screen won't come on when I plug it in is when I have UCH set to off, which is normal. Also, the Wismec updater is connecting to the mod with the screen off. I'm using the latest version of this firmware (53f04f5)

TheGiolly, 2016-11-21:
When the box is in standby and I connect the USB cable it starts charging normally with the indicator on the screen.
If I plug the cable with the box off I have to press the fire button to start the charging process (also confirmed by using ad USB ammeter).
Box: Reuleaux RX2/3
Firmware: 53f04f5

SEEKNDESTROY, 2016-11-21:
My screen comes on even when the mod is off when I plug it in with UCH turned On. I know mine is a 200S...I'm just providing data for @ClockSelect

ClockSelect, 2016-11-21:
Beware you may sometime think the mod is off while it's not.
If you use a screen saver, the mod switches off at the end of the screen saver duration; but if you do not use a screen saver, it stays silently on during 3 minutes before really switching off (3:30 if you're in stealth mode). During those 3 minutes, it will wake up on USB attachement (since it's not really off); after that, It should not wake up any more without a button action.
Also, if "None" is configured as screen saver with a non-zero duration, you may think the box if off while it's not; it's just on with a black screen as long as the configured screen saver duration.

ajape1, 2016-11-21:
I don't know exactly what UCH is, can you please tell me? And I'm glad that it's not something wrong with my unit, I have that exact same thing happening, it turns on from a USB connection immediately while running the Wismec firmware but if it's off and I'm using the MyEvic firmware it does nothing until I press the fire button. :-(

ReMicroN, 2016-11-21:
I confirm that I need to click fire while box is off to start charging. RX2/3.

ajape1, 2016-11-21:
ReMicroN, can you also confirm you don't need to do so while the mod is on the Wismec firmware, please?

drminol, 2016-11-21:
I have the same issue on my eVic VTC Mini. If I manually turn of evic, and connect charger to microusb nothing happen. I must click the fire button, or turn on evic to start charging.

ClockSelect, 2016-11-22:
Revert to stock firmware, wait 5 minutes to be sure, then connect your USB cable. Does it wake up?

ajape1, 2016-11-22:
Yes, it does, The first time I checked after about 40 minutes of it being turned off, since I had to get a shower and make a couple of sandwiches and a shake to go with them, hehehe :) The second time I checked after 5 minutes of it being turned off. On the 4.13 Wismec Firmware, when you plug it in while it's off, the filling battery animation immediately comes on the screen automatically and it starts charging right away.

ajape1, 2016-11-22:
You know I also noticed a couple more things while on the Wismec firmware... The temp readings are HUGELY different, like a hundred degrees Farenheit different I mean. While vaping on TC SS on the same setup, at the same wattage, in MyEvic the temps would only climb to about 260-270 F on a 5 seconds draw. Those same exact draws on the Wismec firmware would see temps rising to 370-380 F !!! That's a HUGE difference in reading! And also, back on the Wismec firmware, I've noticed that I do not get forcibly booted out of TC SS to power mode, the coil readings are locked in and they stay that way, and I stay in TC SS all the time without any problems. Those are two very important differences I thought I'd also let you know about. :)

ClockSelect, 2016-11-22:
@ajape1 it's another subject. See #115.

ClockSelect, 2016-11-22:
Found the USB issue: you just lose the USB wake-up ability when LSL mode is ON.
Restoring this ability would increase power consumption for something not really usefull, IMO.
I'll see that when I'll get a multimter to make consumptions test.

ajape1, 2016-11-23:
Fantastic! Please let us know what your tests show, and if it's something negligible then maybe you can correct this problem in MyEvic? Thanks! :)

ClockSelect, 2016-11-23:
I'm afraid it's not neglictible. I can't tell since I can't really measure yet, but the datasheet of the MCU tells you that PLL clock is not really ascetic.
I have a project to buy a multimeter so I'm watching reviews. Will take a few weeks.

ClockSelect, 2016-11-23:
Well, know: if you plug the USB cable, just click the button. Not so hard as a workaround, IMO.

ajape1, 2016-11-23:
LOL No, not so much of a burden, I agree, hehehe :)