ClockSelect/myevic #188: Behavior of ±1 Watt option when enabled with the power level set to a non-integer

Justin-L-Franks, 2016-11-21:
I noticed that if you turn on the ±1 watt option when the power is set to a non-integer, the decimal is kept when adjusting the power. For example, if the power level is set to 30.3 watts when the ±1 watt option is enabled, adjusting the power level up will go to 31.3 watts, 32.3 watts, and so forth.

I think that this option is meant for the people who want their power level to be a non-integer, so it would make more sense to change the behavior of this option to switch to the next-nearest integer the first time the power level is adjusted. So when the option is turned on while the power level is set to 30.3 watts, increasing the power level will first go to 31 watts, then 32 watts, etc., and decreasing the power would go to 30 watts, 29 watts, and so on.

**Box Model:** Cuboid v1.01

ClockSelect, 2016-12-15:
You should no longer be able to select a non-zero decimal value of power when 1Watt is selected.

Justin-L-Franks, 2016-12-21:
This is working perfectly. I tested setting various non-integer power levels then turning on the ±1 watt option, and it automatically sets the power to the floor of the whatever was set (40.2 W -> 40.0 W, 50.9 W -> 50 W, etc.). Thanks so much!

closed 2016-12-21 by ClockSelect