ClockSelect/myevic #189: How to make own custom firmware

LockManipulator, 2016-11-21:
I couldn't see a way to message so sorry if this shouldn't go here. How would I make my own firmware? Googling it only brings up already made firmware or guides on how to use nfirmwareeditor.

ClockSelect, 2016-11-22:
What do you mean really by "making my own firmware"?

LockManipulator, 2016-11-22:
I mean things such as this project. Adding features such as a clock, games, etc. On NFirmwareEditor, it really only allows changing fonts and drawing your own characters. But if I want to make my own firmware with different positioning of items on the screen, different menu, and other things how would I do so? I have some background in coding but don't know how to transition that to this.

ClockSelect, 2016-11-22:
At the bottom of the readme file, there is a "build" section that let you know how to clone and build your own copy of the firmware; there's also some issues related to this in the issue base if you get into troubles. If you learned some C code in the past, it shouldn't take you long to find your path.

LockManipulator, 2016-11-22:
Thanks! Exactly what I was looking for, can't believe I missed it there.

closed 2016-11-22 by LockManipulator