ClockSelect/myevic #191: Am I allowed to translate myevic?

thomaskunrath, 2016-11-22:
Hi all! I'm not having issues at all! Firstly I must say everyone is loving your custom fw! But the thing is, I had to translate the whole page so people could use it and understand what every option do! Obviously, I wrote that all credits are yours! But, looking forward in order to keep helping people here, I was wondering if is possible to translate the whole Fw, or just some parts of it. Translate all is not necessary cause many options speak for themselves. Looking forward for your answer, and congratulations one more time, Thomas. 
**Box Model:**
Rx2/3 + Aromamizer Supreme 
**Coil setup:**
Dual Clapton 26/34

ClockSelect, 2016-11-22:
Well.. Sure you can do whatever you want, since you can pull your own fork for your own purpose.
But remember you only have 64 pixels per line, and that memory is limited to store language-specific glyphs. If, say, "Coil", translates in your language into "Kwøzwićzÿ", you may soon get into troubles.
The main quality of English is that it's *compact*.

bluenazgul, 2016-11-22:
@thomaskunrath i think the best would be, if you only translate the "README.md" like i did with the "README_de.md" and keep this file Updated.

thomaskunrath, 2016-11-22:
I already translated README.md to Brazilian Portuguese! I thought in translate the FW just to make thinks easier, much more people would use like this!

ClockSelect, 2016-11-23:
@thomaskunrath You did? It's awesome!
Best way is by github fork/pull request process, but if you really don't master those tools, just give me a link to your .doc file and I'll push it on next release.

ClockSelect, 2016-11-23:
@thomaskunrath You'll convert it to .md when you'll take time to see how it works (.md is basically a text file) but even just having a .doc in brasilian is awesome.

thomaskunrath, 2016-11-23:
Yes mate, it's done. I've sent you by e-mail! Hope you like!

thomaskunrath, 2016-11-23:
Hi there!

 

It’s Thomas, from the little chat of github.

 

Honestly, I don’t really know how to upload a formatted text file to github, or fork/pull request process.

 

I’m not a dev, so I could mess up something! 

 

The .doc file follows attached! Hope you like. Is BIT different, concerning installation procedure. I just put some images to make things easier for the people here in Brazil. Most of them have no way to understand English. So, fell free to remove them if you want.

 

In need any changes, let me know.

 

Oh, I almost forgot.

 

The “expert” part, as you wrote in the .md original file says that most people won’t need, so, I just translated a part that says “don’t bother with it, but also, don’t play around just do see “where I can go changing this options””.

 

And, I have only one doubt about the FW. In the Expert Menu, over Bat Submenu, is possible to change the Model of the battery I’m using. Changing this would make any effect?

 

Also, water mark of the document is just of my FB Vaping Community, so when you copy the text, it won’t go together!

 

Best Regards, and, thanks one more time for this awesome FW,

 

Thomas Campagna Kunrath

ClockSelect, 2016-11-25:
@thomaskunrath I'm afraid I didn't.

thomaskunrath, 2016-11-25:
@ClockSelect here it go! If you want me to change anything, let me know! 

[Introdução a instalação da Custom FW.docx](https://github.com/ClockSelect/myevic/files/613492/Introducao.a.instalacao.da.Custom.FW.docx)

ClockSelect, 2016-11-26:
Since I can't read portuguese, I can't really see if there are things to change or not^^
Let's wait and see what Brazilian users will tell us.

thomaskunrath, 2016-11-26:
I will finish the "Expert" part today. @ClockSelect 

There's some paragraphs that I changed a little bit to sound more understandable in Portuguese. Just translating straight away would be a little confusing!