ClockSelect/myevic #193: SS316L temp coef

kablek, 2016-11-22:
Hi!

I just tried TCR for SS, it apears default TCR for SS is waaaaay of (120) from other people on the internet I have found out it should be around 92.

Here is the source of my TCR values: https://www.reddit.com/r/electronic_cigarette/comments/3xq9qr/vtc_mini_tcr_values/

t2dYuri, 2016-11-22:
SS316 TCR in the latest native Joyetech firmwares = 120. In myEvic you can change it to any value, depending on your wire.
I'm testing all my new wires with a dry cotton and adjusting TCR to start changing it's color at 210 degrees (because i want all my coils to work with the constant temperature limit)...
So all my wires with label "SS-316L" have stickers with measured TCR from 110 (no-name 0.30mm), 115 (few british wires 0.25, 0.35, 0.40mm), 130 (green UD 0.32mm)

RenegadeTitan, 2016-11-22:
this information has been very difficult to find accurate publications for due to  SS316(L) having different properties/TCR values due to variances in purity. but here are some rough figures:

stainless steel does not have a consistent TCR at various temperatures unlike Ni or Ti mostly do. it changes as SS heats up. its TCR~92 at around 70 degrees F and TCR~130 at ~400 degrees F (roughly). 120 is typically used because it is roughly the average of the vape temperature range. other firmwares use different TCR to compensate for inaccuracies in their other readings (base ohm etc).

enricorov, 2016-11-22:
I use 098 for my UD SS316L wire. It's up to one's taste, really.

RenegadeTitan, 2016-11-22:
all the TCR does is adjust the accuracy of displayed temperatures therefore affecting the interpretation of your set temperature.

SEEKNDESTROY, 2016-11-22:
In my opinion, the TCR value changes depending on the type of build, gauge diameter etc.
[Steam Engine's Wire wizard](http://www.steam-engine.org/wirewiz.asp) I'm not sure how accurate it is but maybe someone else will find it useful.

enricorov, 2016-11-23:
@seekndestroy I found steamengine to be quite accurate in the past. 

P.S. I'm not that impressed with Hardwired, which is a shame.

ClockSelect, 2016-11-23:
You're all right, SS TCR depends strongly on build and gauge.
I use the same method as @t2dYuri to calibrate my TCR (dry cotton test). Pluging the coil in boiling water (guaranteed 100°C) is another way. And TCR may vary from one build to another around 100±20.

kablek, 2016-11-24:
Okay, reading responses I have actually remembered some stuff I didn't take into account and I am aware of, SS316L is type of stainless steel and yes I guess just like the pure iron (Fe) it probably does not have constant coefficient of resistance, but I think it shouldn't vary too much in material itself since SS316L is specific type of SS with known mixture of other metals.

Here is the source showing what exactly is inside SS316L: http://www.azom.com/article.aspx?ArticleID=2382 

Here is graph stolen from some subjects I am taking at university (I study electrical engineering and this graph is from electrical engineering materials subject) 
[image]
Graph shows specific resistance in relation to temperature, as we can se for iron (Fe) it is not linear function.

Maybe I can get some people at the university interested and make accurate scientific measurements of different SS316 coils at temperature range from about 20°C to about 400-500°C, if the function how resistance changes is not linear maybe we could implement actual compensation of that. 

But to make things safe and sure it would be nice to test different SS 316L manufacturers and see how consistent data is. Currently I only have one type of SS 316L and I will try making some measurements and I will comeback here with some actual data!

ClockSelect, 2016-11-25:
That's interesting.
Even if TCR is not constant over the temperature range, by taking the resistance slope on the middle of your target temp range (usualy 20°-220°, so say you take the TCR at 120°C), you get a fairly good approximation that's good enough for the purpose of vaping.

tharealgc, 2016-12-04:
For RX 2/3 - 316L TCR is perfect at 117-118...

ClockSelect, 2017-01-24:
@kablek No news?

closed 2017-01-26 by ClockSelect