ClockSelect/myevic #194: [Suggestion] Battery voltage cut-off

mindtrapper, 2016-11-23:
Would it be possible to add a setting for lowering the voltage cut-off? DNA mods have such an option and can be set to bring the battery down to somewhere around 2.8V. Factoring in some extra sag during firing this should be safe for our modern 18650s. I think it will be a pleasant boost in autonomy.

Daveee10, 2016-11-23:
The cut-off voltage managed by the Battery profiles... You can see in the source, how the Cut-off voltage is 2,8V for every battery included the Generic...

ClockSelect, 2016-11-25:
The cut-off voltage of the custom battery profile can be modified whane edited with the NFE battery profile editor.

mindtrapper, 2016-11-25:
Doesn't seem to be doing anything. I edit the values for Custom profile and upload settings to device but then nothing changes in device. Battery in device settings still seems to be 30Q, the one I had selected before making changes with NFE. There is no "custom" option there, just some batteries and generic. 

I did some other changes in NFE too, just to see if they will upload to device (like changing mode, watts, etc) and nothing changes. Also in battery profile editor the lowest for 0% is 3V and cut-off 2.75V. Is 0% actual battery voltage and cut-off the battery voltage when firing (sag)? I haven't emptied a battery with the mod yet so I can measure it externally, but it seems to be same as stock firmware.

ClockSelect, 2016-11-26:
You have to set the battery model to "CUS" (two seconds fire click when configuring battery model) to activate the custom profile (see README).

MoDuLah, 2016-11-27:
Still trying hard to find that battery profile editor... hm...

closed 2016-11-29 by ClockSelect