ClockSelect/myevic #197: Manually change puff counter?

dexty32, 2016-11-27:
Is there a way to manually change your puff counter? I like to keep track how many puffs i have total, but after i did my first update, it reset it, id like to manually add it back up now, if thats possible?

evic VTC mini.

kcypher, 2016-11-28:
Hello, I can do this for you, if you want!

dexty32, 2016-11-28:
That would be great! how would it work, will you give me a download link or something? If you can, set it at 26k puffs.

kcypher, 2016-11-28:
Wich FW do you use? The newest myevic from this repo? I would make a new menu to change manually the puffcounter. I would send it to your email, if this would be no problem for you.

dexty32, 2016-11-28:
Yeah, the newest one, and e-mail would be great! Thanks for helping by the way for not that important or an issue.

kcypher, 2016-11-28:
I have still one question: Wich Box do you have? Maybe this could be good to know, haha!

dexty32, 2016-11-28:
evic vtc Mini :D

kcypher, 2016-11-28:
Ok, give me some time. I tell you when i'm done. Maybe 1h or so. ;-)

kcypher, 2016-11-28:
Ok, finished. Whats your mail?

dexty32, 2016-11-29:
Oops, I was already sleeping. My email is frankerik32@gmail.com

Thanks for the help :)

kcypher, 2016-11-29:
Ok, sent it to you. In the Vapingmenu there is now a new Submenu called PuffCount. Here you can manually set the PuffCounter. 
Pls let me know if everything works just fine, or not.

dexty32, 2016-11-29:
Thank you! I'll try it later tonight and I'll let you know if it works.

kcypher, 2016-11-30:
Hello, do you tested it?

ClockSelect, 2016-12-01:
That's exactly the way I like seeing people interacting on this firmware. Thanks, guys.

dexty32, 2016-12-01:
Sorry, took a while to reply. But it works great kcypher, thank you so much! Takes a little while for the scrolling but wont be doing this again for a while anyway.

Have a great day!

dexty32, 2016-12-01:
Hey, i have one last question, i was was using one of the older myevic firmwares, now i installed the one kcypher gave me, i assume its newest one, the wattage now jumps down and up realtime, aswell as the ohms, my ni80 coils are going from .218 to .240 when firing, is that normal? Thanks.

kcypher, 2016-12-01:
I took the newest one, yes. I'm not so familiar with temp-vaping. The only thing i know is that you have to lock the coil resistance. But i'm sure that you've done this already! However, it has nothing to do with the PuffCounter, don't worry! :-D

closed 2016-12-01 by dexty32