ClockSelect/myevic #198: how to convert bitmaps

kcypher, 2016-11-28:
Hi folks,

how do i have to convert bmp's for fonts.c?
The VTC Font Editor and the convertion tool from the sdk have different byte orders like data in fonts.c. Witch tool do you use?

Thank you in advance! Kay

Oh yeah, my box is a vtwo mini with an lemo3. Coil: 0,2Ohm (dual)
Power: 60W

ClockSelect, 2016-11-29:
I use the NFirmwareEditor or VTCFont to modify pics/convert BMPs into the firmware.
Then I use a little home-made command line to generate the fonts.c file.

kcypher, 2016-11-29:
Cool, is it possible get your cmd line tool? Did you use gcc for that?

ClockSelect, 2016-11-30:
You may download it [here.](https://www.dropbox.com/s/lk1g0vhqariy3jr/expfont.c?dl=1)
It requires an uncoded firmware as input.

kcypher, 2016-11-30:
Thank You very much. So i have to change the font from an standard firmware, and then extract the fonts.c from it with this tool? Is that right?
I also found an very easy to use tool. Its good if you only want to insert one or two bitmaps. http://www.holger-klabunde.de/mysoftware/bmp2c.zip
Usage: bmp2c pic.bmp v
In this example it would generate a pic.c and a pic.h file. The option v is for vertical byte order, the one we have. I think its a very usable tool for evic-sdk.

ClockSelect, 2016-12-01:
Everybody makes its own tools, and I can't judge them all. I give you mine, and that's the best I can do.
The font used in my firmware is a vertical byte ordered font, if you wish to call it so. The second font (block two, horizontal) is not used at all.

kcypher, 2016-12-01:
ok, thank for the information.

closed 2016-12-01 by kcypher