ClockSelect/myevic #199: suggestion: limit in powermode

xabbu16, 2016-11-30:
Dear all,

could it be a good idea to include a new feature to customize the box for new beginners to minimize potential miss-use / errors: 

My procedure is to deactivate all modes but the powermode.

A possible suggestion is to integrate the feature to delimit the power range between to values (9-30) so that a absolute beginner / newbie to the vaping can not damage the coils?

Regards,

Chris

**Box Model:**

**Coil setup:**

dywersant, 2016-11-30:
This software in any shape or form is not meant for beginners, or at least i see it that way.
In order to protect a new user you want to restrict them only to tc mode, and supply matching prebuilt coils.

ClockSelect, 2016-12-01:
It was the original prupose of the "SMART/START" mode. It has not reached its target popularity, if I recall correctly.
VTC and all of Joyetech-like chipset boxes are definitely not beginner boxes, and I'd swear that anybody lurking at this site is also definitely *not* a beginner.
Maybe the best answer to this issue would be to add such a statement line in the readme file.

closed 2016-12-15 by ClockSelect