ClockSelect/myevic #202: Display mAh

arafey, 2016-12-02:
It would be useful to view the amp-hours used since last charge in order to monitor battery usage or to diagnose a failing/old battery with a low charge capacity. 

It should be theoretically possible to have this since we already can see the "on" time in the puff counter as well as the instantaneous battery output current.

arafey, 2016-12-02:
Also, I'm assuming if this option is implemented, it could be located on the third line (where the puff counter, mL/day, etc. is), or as an alternative to battery percentage or voltage. The latter option would require a preset and/or a user-adjustable "battery capacity" value (probably between 1000-4000 is a good range) in order to decrement the battery bar at an appropriate rate.

easyyu, 2016-12-02:
This option would be really nice...

MicroSur, 2016-12-05:
There are no sensor to get real current from battery itself.

arafey, 2016-12-05:
Wouldn't Ohm's Law be sufficient? We also don't have sensors to measure mL of liquid but that option still exists.

MicroSur, 2016-12-05:
If you store efficiency of all cirquit boards for each mods and for different powers...

arafey, 2016-12-05:
Without that efficiency information, you would probably have up to 15% error in the reading (assuming most boards are at least 85% efficient), but that's still valuable. If you include a user-adjustable offset (like mL/kJ option for the mL display) then users can calibrate it to remove that error. My mL reading had 50% error originally, yet that option still exists.

ClockSelect, 2016-12-11:
That's doable, but since your charger gives you the mAh when recharging the batteries, what's the point to have it on your box?
Regarding built-in batteries, problem lies in box efficiency that's not really measurable.

arafey, 2016-12-15:
You're right, most chargers do give mAh and I didn't consider internal batteries. Some people also use the USB port to charge. 

I still think it's worth eventually adding but it should be low priority.

ClockSelect, 2016-12-15:
Trick: if you set your mL/kJ ratio to 100, the liquid displayed will in fact be total energy dissipated in tenth of KiloJoules (Joules x 10e-4). Multiply by 10000/3.6 and you have the mAhV (milliAmp-hour-Volts) value. Divide by your mean voltage (say, 3.6 for a generic battery) and you have a rough value of your mAh.

closed 2016-12-21 by ClockSelect