ClockSelect/myevic #203: Clock on Cuboid default year

Gremdude, 2016-12-03:
First off, I love the Myevic firmware for the Joyetech and Wismec devices. The default firmware on the Cuboid is pretty terrible--especially the current version 5.14  with its unreadable ohms, volts etc. Also the insane speed that the temp and watts scroll is just ridiculous and have no idea why Joyetech doesn't fix it! Now onto the issue. For the clock, I understand that the Cuboid does not have an RTC battery so is not going to keep the time when you remove the battery. The problem is that the default year when I go to reset it something silly like 2068 . Soooo, I have to scroll ALLL the way back to 2016 when II change the batteries which kind of makes me want to always charge the batteries in the mod which I don't prefer. Any way to set the default year to something reasonable like 2016 or how about 2017 now as we are almost there! Thanks again for the great work you are doing here!

ClockSelect, 2016-12-06:
I'm aware of this.
Problem is that first readings of the clock are off even if it's saved.
This is not so simple.

Villista, 2016-12-07:
#179

GravityRoller, 2017-04-17:
Gremdude,

v3.14 Firmware on CUBOID has the old display (preferred i.m.h.o., as much easier to read) and includes the rest of the other updates (Preheat, etc.) of v5.14

BigBadBlackWolf, 2017-04-18:
ClockSelect is probably no longer available for the further development /
improvement - why ever. MiroSur has apparently accepted the case. For more
information see:
https://github.com/MicroSur/myevic/find/master?q=
Last current version of myevic is 3.30 from 09-04-2017. With Flappy Bird
and Tetris.