ClockSelect/myevic #204: "Real" lock for coil resistance

dabugen, 2016-12-05:
Hi,

is there a way to do a "real" hard lock of the coil resistance in the firmware? Especially during the cold days here and if being outside and having the device in stand-by for a longer time, I find that the locked resistance, even though it was locked, is being changed, because the difference compared to before seems to be to big. This is really annoying. Is there a way to have a real lock of the resistance that will NEVER change until I change it or unlock it?

Thank you :)

ClockSelect, 2016-12-06:
The best way is to *not* lock it.
If you don't lock the resistance, it'll ask you if it's a new coil or not; say no and it'll keep its value.

dabugen, 2016-12-06:
I see, good trick, but is there no way to simply make the lock a lock? I mean currently, the lock is no lock at all, make no sense to me.

MicroSur, 2016-12-10:
Lock resistance is very useful, I want vape, but not answer mod`s question )
Bug with changing locked res down, but never up - fixed easily.

dabugen, 2016-12-10:
My "locked" resistance changes into each direction, it just sucks. Please make it a REAL lock.

I fear though that this project might be dead, no more changes for 14 days, that is very uncommon....