ClockSelect/myevic #205: Clock not saving

AlpinaX, 2016-12-08:
**Box Model:** RX200S

**Coil setup:**

Hi,

First off, great firmware! Much better then to stock firmware for my RX200S.

Issue I'm having is that when I set the time or date, and hold the fire button to save, it's not saving and reverts back to 00:00:00. 
I understand that with my particular mod, when taking out the batteries it will reset to default, but I don't mind setting it up everytime I reinsert the batteries.

AlpinaX, 2016-12-08:
Issue seemed to have fixed itself when uploading a new Logo onto the mod. It is now saving the time and date I set.

Mod can close.

ClockSelect, 2016-12-11:
Strange.
Anyway, Clock should not be taken as reliable on boxes that do not have a clock in the first place. I may work fine or not, milleage vary from box to box even in the same model.

sidran, 2016-12-11:
AlpinaX, I am using Evic VTC mini so I am not sure if this would work for you.

I once got into 00:00:00 situation and FWUpdater.exe of [VTCFont package](https://www.dropbox.com/s/83zd19gu05pl3r6/EvicVTCFont.rar?dl=1l) did help me. Try using "Reset mod" in expert settings. Also, same menu allows you to sync device clock to your computer clock, with one click.

closed 2016-12-15 by ClockSelect