ClockSelect/myevic #206: Power Curve troubleshooting (add a limit to nullify every +X value)

Derducken, 2016-12-09:
So, after playing for a lot of time with the Power Curves feature, I feel like I did when you first added it:

* It's still awesome if you vape like I do, making vaping with Kanthal feel like you have some control over its temperature.
* It **does** need the "limit" feature I had mentioned before.

Now, I'll explain the "how's" and "why's" of what I call "a limit".

If you look at the typical PreHeat function for Power Mode, apart from the amount of power and the time a kind of "Boost" will be applied to the "main power value", you can also set a delay. This delay is useful when chain vaping, 'cause instead of giving your coil another boost, and another, and another, each time you press the fire button, possibly ending up vaping more frequently at your "boost", your "preheat" value instead of your actual value.

Now, let's turn to the Power Curves.

I've ended up using the following "Power Curve" with at least three of my coils: I start at 120% for 0.5secs, go to 110% for another 0.5secs, then to 100% for another two half-seconds. Then I drop to 80% and raise the value by 1%, half-by-half second.

What this "feels" (and, I guess, works) like, is this: you have the equivalent of "a starting boost", a "preheat", for 1 second: half a second at 120%, another half at 110%. Then you vape "normally" for 1 sec. Then, the reason for the power drop, is that the coil starts getting too hot if you keep vaping for more than 3-4 seconds. By dropping the power level you allow it to cool down a bit and equalize. You raise the power until the end to keep the temperature relatively "steady".

I don't think the above would work with ANY coil, and every user can try to experiment to create a similar experience with his own setup. But let's go to the problem itself.

Since I'm using the "over 100% values" as a "PreHeat alternative", and I guess that's how most people would use the feature (a PreHeat is more useful when actually vaping compared to, say... a mid-Heat :-D ), when I chain vape, each time I hit the fire button ALL those values are re-applied in sequence. Since it's not a Boost, it's a full power curve.

Exactly like PreHeat can make a coil heat more than you like, the same thing happens with the power curves. Only solution, to forget the starting "preheat" and create a different curve, starting from 100% and dropping.

Except...

Except if there was a kind of "limiter", EXACTLY like the way the PreHeat works in Power mode, where if you pressed fire again during an X seconds window, "every value over Y wouldn't be applied". You could hardcode the value to be 100%, since not many people would tweak it, but, heck, if it was editable itself it would be even more awesome.

Of course I know I'm nitpicking but, hey, a man can dream :-)

ClockSelect, 2016-12-09:
Maybe for simplicity we could just apply the preheat delay parameter to the curve?

Derducken, 2016-12-09:
Of Course! The only thing that matters is to be able to not re-apply "anything over 100% for X seconds" to avoid burning the cotton.

ClockSelect, 2016-12-11:
Let's try it this way

StepWi, 2016-12-12:
It is ok for me (Cuboid Mini).

closed 2016-12-15 by ClockSelect