ClockSelect/myevic #207: Auto TC calibration

Derducken, 2016-12-09:
Here's a silly idea - and the way it could be implemented: auto-calibration for any kind of (TC-compatible and "within the reading limits of the mod") wire. Know how even the same type of wire can have differences based on brand? How you may need to adjust the TCR for each of your wires manually? This is the "automagic" version. While burning cottonz!

We could have a "Monitor and auto-set" in the TCR menu, asking the user to do three cotton burn tests. The mod would monitor each clicks duration and resistance change to "extrapolate" the TCR value.

It wouldn't be as good as manually tweaking the TCR, or having a true "material curve", but it **could** help a lot when, say, trying to adjust the curve when using coils made from mixed materials (like Kanthal + SS). Instead of toying around with numbers (and failing), you could "auto-tune" a "baseline" for your TCR this way and then go from there.

ClockSelect, 2016-12-11:
I tried to do some calibration stuff a wile ago, but it's complicated and doesn't help much.
Manual methods work fine and I think we'll stick to them before some new enhancement in the hardware.

Derducken, 2016-12-11:
Kays, just throwing ideas here. Shall I close this issue or will you?