ClockSelect/myevic #208: Sweet algorithm for TC  burns cotton

roaminghawk, 2016-12-10:
Hi,

eVic VTC mini here.
When using the Sweet algorithm for temperature control and testing a dry piece of cotton it burns the cotton and shoots beyond the set temperature.
Is there an option I'm missing that prevents cotton being singed? As this defeats the purpose of TC.

Thanks.

Derducken, 2016-12-10:
Yes, I saw that too now that I'm toying around with TC - I was only using Kanthal before.

Here's a way to replicate: feed a strip of cotton through SS316L coil. Lock resistance, set temp and Watts, set algorithm to Sweet.

Now, if you press and HOLD fire, you'll see first a smaller "pulse", where temp is onviously increasing (the coil glows a dull red) only to then back off, stay low, and then, after some seconds (sorries, didn't count), the coil pulses once more to a way brighter red that burns the wick before backing off again.

I didn't try extending the limit of each hit to 15-20 secs to see if it pulses a third or fourth time 'cause I'm using relatively old batteries and don't want to push them too much.

ClockSelect, 2016-12-11:
Sweet algo differ from factory algo only after temperature stabilization; you should not experiment a different overshoot of the set temp between the two. Theorically, you'd have the same result concercing cotton burn with the guenuine algo.

Derducken, 2016-12-11:
Nope, that's the thing, when I did the test with "no algo" the cotton didn't burn. Discolored, yes, but not an outright black-colored burn.

I know that under *typical* use there (probably) wouldn't be any problems, but when adjusting the TCR of a "strange" (as in "not-in-the-default-values") wire, you should either use no algo or stop pressing the button after the first "pulse" is noticed.

ClockSelect, 2016-12-11:
Algos may work better or worse than the default depending on your setting; that's why the manual states that there's no "best" algo. It always depend on what you have.
I suggest you make dry burns with the original algo to set up your TCR (or enen use some other method, such as plunging the coil in boiling water while reading the resistance), then switch to one of the advanced algos depending on your needs and feels.

Derducken, 2016-12-11:
A-ha, okays, thanks for clearing that up.

roaminghawk, 2016-12-11:
As Derducken stated, the original algo don't burn the cotton nor over shoots over the set temperature. 
I first noticed it when I forgot to open the juice flow control after filling the tank, and I started vaping, after a while I noticed a burnt taste but I ignored it since I'm on TC and that can't happen. A few more puffs and it was evident that the cotton was badly singed.
Reason I'm reporting this instead of just using the original algo is that I like the "sweet" algo very much more than the original.

ClockSelect, 2016-12-15:
Maybe it's just that the sweet algo is not thruly adapted to your setup; we should look at monitoring curves to have more insight on this. Or did you try the PID algo? It often works better when others fail.

mattjanus, 2017-01-13:
SS316 isn't a very good wire for TC. I would suggest using a wire with a better curve. NI200 is the most accurate afaik. If you want to stick with stainless I would suggest trying ss430. It's got a slightly better curve. I've used 430 for 6+ months now and haven't looked back.