ClockSelect/myevic #209: Latest binary ([#206] Preheat Delay applies to power curves) is completely nonfunctional on my Cubiod v1.01

Justin-L-Franks, 2016-12-14:
**Box Model:** Cuboid (original 2-cell model) v1.01

I was interested in trying out the latest commit, where the preheat delay also applies to the power curve feature. Unfortunately, upon uploading the latest binary, "[#206] Preheat Delay applies to power curves", my Cuboid went completely dead and would not turn on. I hooked it back up to my Mac with Joyetech's firmware update tool and it showed up as firmware version 0.00.

Luckily, I was able to revert back to the second most recent commit, "[#191] Brazilian Portuguese Documentation". After doing so, everything came back and is now working properly.

I tried uploading the newest firmware binary once more and the same thing happened.

Any idea on what could be causing this? The power curve delay feature would be quite nice to have, so I was disappointed when the binary which included this feature refused to run on my Cuboid.

Thanks so much!

StepWi, 2016-12-14:
I have installed it successfully on evic basic, cuboid mini, cuboid 1.01 and 1.02, from Windows with the uploader from NFirmwareditor. 
I have sometimes the same problem ou have described, but normally firmware installs in the second trial. Have you checked if your downloaded file #206 is broken / incomplete?

ClockSelect, 2016-12-15:
Sometimes, the file may be badly downloaded; it's generally what causes this kind of issues.

ClockSelect, 2016-12-17:
No news, good news.

closed 2016-12-17 by ClockSelect

Justin-L-Franks, 2016-12-18:
Still no luck, I tried downloading it several times, using a different known-good USB cable, tried booting into Windows (I have a Mac but have a Windows partition), and tried a different computer (a Windows laptop). The same problem occurs, and I have to go back to the previous version.

Justin-L-Franks, 2016-12-21:
Alright, so I was able to get the latest binary "[#217] Reset counter Estimate of vape velocity in ml or ml/day (experimental)" uploaded to my Cuboid and it is working. For whatever reason, #206 never worked at all for me.

ClockSelect, 2016-12-21:
Great!
Maybe It was a proxy cache issue with #206. What country are you downloading to?