ClockSelect/myevic #211: "Atomizer Low" / "Atomizer Short"

dabugen, 2016-12-15:
I would like to vape my 0.06 NI200 build on my Presa TC 100 (am using that build fine with other non-Joyetech mods), however, the Presa TC 100 with the MyEvic firmware tells me "Atomizer short" once I fire. Before I do that, the resistance is reads absolutely fine and says 0.06, which is absolutely right. As soon as I fire, it switches to 0 and says "atomizer short". I guess there is a limit in the firmware to not allow builds < 0.1 ohm, right? Because with a 0.12 Ni200 build, I can just fire fine and it works perfectly. Can that limit be removed please?

ClockSelect, 2016-12-15:
Rez showing zero means the box resets when you try to fire, I could mean it goes below the 2.8V hardware limit.
Doses it behaves this way with a fully charged battery?
It may happen when you use the BVO feature to positively offset the battery voltage, and batteries runs out of power. Try to set BVO to zero to check.
The built-in limit for TC is 0.05Ω, so it should be fine.
Or maybe an issue with the coil?

dabugen, 2016-12-15:
I´ve just set the PWR in TC mode to 50 watt (absolutely no problem for a 26650 and it is freshly charged as well) and just 180°C. Resistance shows 0.066. As soon as I fire, "Atomizer Short". If I put the atomizer to another mod, it is all fine, no issues (also shows 0.066 there) and it is using the same battery actually. BVO is not set to anything (0). So not sure what else I should do, I can´t get the "Atomizer Short" to go away. As soon as I take out the battery and put it back in, it is all back to the right values (0.066). But firing just once makes show "atomizer short" and sets the resistance to 0.00. Not sure what I might be missing?

ClockSelect, 2016-12-15:
May be the algo?
Or TC max pwr limit?
(Trying to guess, too)

dabugen, 2016-12-15:
Algo is the standard one (Off). TC Max PWR Limit at 50W? Make no sense... I fire my Titan coils easily with 80W as starting power in TC mode. If I put on a NI200 wick with 0.12 ohm, it is working fine. Anything below 0.1 ohm seems to give "Atomizer short",  in TC mode!

ClockSelect, 2016-12-15:
Does the COM terminal says something?
Does the box actualy resets when firing or not?

SEEKNDESTROY, 2016-12-15:
I had a similar problem a month ago with a 0.08 and 0.09 316L SS 24 Gauge Parallel Dual coil builds, in SS TC mode. I kept getting the same message as @dabugen... the only way I was able to vape in SS TC mode was by removing my RDTA and disabling all the modes but the SS TC mode... and even then it jumped back in to wattage mode from time to time even though wattage mode was disabled. I'm now doing 0.1 or above builds and all is well but I wish I was able to run my build at 0.08.

That's pretty much all I remember... I hope this helps.

Mod: RX200S
Build: 316L SS 24 Gauge Parallel Dual coil

ClockSelect, 2016-12-15:
I'm not sure it's the same issue. If you fall back to power mode with heavy SS builds, you may try to disable the "Check" option in the coils menu (see documentation).

SEEKNDESTROY, 2016-12-15:
I had and still have "Check" disabled. Having thought about it some more, mine said "Atomizer Low" and not short. I shouldn't have any problems at 0.08 or 0.09 in SS TC mode, right?

ClockSelect, 2016-12-15:
You get a Low warning after you felt back to power mode with a rez < 0.1Ω; are you sure you disabled the check at the time you got the Low warning?

SEEKNDESTROY, 2016-12-15:
I'm 90% positive that it was disabled but it's been so long that I can't be 100% sure.
Here is what I remember, after building and wicking I screwed my RDTA back on my RX200S in SS TC mode and was instantly kicked out of TC and in wattage mode. The resistance display was flashing...When I set it back to SS TC it wouldn't let me, it stayed in wattage mode, never once did it ask me if the coils was new or the same. Only way around it was to unscrew the RDTA and disable everything but SS TC, as I mentioned before. That's why I'm 90% sure "Check" was disabled.

I guess I just have to put another 0.08 build on it to see what exactly is going on. I'm gonna test it either tomorrow or the beginning of next week. I'll let you know.

dabugen, 2016-12-15:
I´ve made a video about the issue here: 

https://www.youtube.com/watch?v=dP-dsP86Vuk

P.S.: Any build > 0.1 ohm works just perfectly fine (without changing any settings, just screwing on another atomizer or removing one of the dual coils, so that resistance is 0.13 ohm then.

ClockSelect, 2016-12-16:
Thanks for the video. I think I see the problem, in your case.
Below 0.12Ω, the firmware limits the power to enforece the 25A limit, but it may exceed it a little anyway (say 25.1A) and the box considers the atomizer short. I increased a bit the error margin for the test.
Unfortunately I can't test it since I've no such low ohms builds; try and tell me.

dabugen, 2016-12-16:
Thanks for your efforts, I just flashed it, but still exactly the same behavior :(

ClockSelect, 2016-12-17:
Could you try to use the COM terminal of the NFE 5.1 and see what it says when this shortage occur?
It would be informative. (Beware, NFE 6+ do not have COM term any more).

ClockSelect, 2016-12-21:
?

ClockSelect, 2017-01-25:
@dabugen Give it another try.

SEEKNDESTROY, 2017-01-25:
@ClockSelect Strange thing happened last with FM [5682f6f](https://github.com/ClockSelect/myevic/commit/5682f6ff5812da829537fa122efae89192542961) I got "Atomizer Short" after turning my Mod off and on after a few seconds.
Here is what happened and the steps I took trying to fix the problem

1. Everything was fine until: I turned the Mod off and back on after about 5 seconds **>** _"Atomizer Short" message_
2. Pushed Power button without RDTA **>** _"Atomizer Short" message_
3. Screwed RDTA back on **>** _"Atomizer Short" message_
4. Turned Mod off and on **>** _"Atomizer Short" message_
5. Turned Mod off and removed one battery, waited 30 seconds and put the batter back in, turned on Mod **>** _Problem Solved_

I since have checked and cleaned all connections in my RX and RDTA (They were clean), tightened my post screws one more time (Didn't seem lose), I have also updated to the latest FM [f1b13a9](https://github.com/ClockSelect/myevic/commit/f1b13a9774e693aa0c1b5292d7763a327f3bf427)

I hope this helps in any way... or maybe it was just a fluke.

MOD: RX200S
Atty: Limitless Plus RDTA, Dual 24 Gauge SS 316l, 0.308Ω build.

ClockSelect, 2017-01-25:
@SEEKNDESTROY Analyzind an issue on a non-reproducible sigle event is hard. As you say, it may just be a fluke.
A new clean resistance reading is performed when box switches on but I can't really see what would interfer with this process.

SEEKNDESTROY, 2017-01-25:
@ClockSelect I figured as much. Thought I put it out there to have a record of it in case it happens to someone else.

Thanks for everything you do... I greatly appreciate it.

dabugen, 2017-01-25:
I can´t test it right now, I´ve thrown away the low ohm wicks and use titanium only now with a resistance of ~0.7 ohm, that just works perfectly fine. But thanks for your efforts, I hope someone else can verify the fix.

ClockSelect, 2017-01-26:
I let the issue opened just in case.
I'll close it in a few weeks if nobody complains.