ClockSelect/myevic #212: Disable Inverted Display on some Screens

sandboxgangster, 2016-12-15:
**EVIC VTC Dual**

Hello, please Disable the Inverted screen on Blank screens, Load Animation, Screensaver off Mode, it’s  Display Always on this ist not good for the OLED.

mfg SB

drminol, 2016-12-18:
Why? It's an option. Who told you that you must use it? Is enyone who? If you crash your screen, it's your problem. Cheers. :)

drminol, 2016-12-18:
And one more thing. It's not bad for the OLED screen, but only for the battery.

sandboxgangster, 2016-12-19:
Warum so Zickig ?
Es ist ein Vorschlag, sinnvolle Verbesserung um der Alterung vorzubeugen.

Zitat von Wikipedia (Translate from the German Wikipedia):

https://de.wikipedia.org/wiki/Organische_Leuchtdiode#Nachteile

Das größte technische Problem stellt die vergleichsweise geringe Lebensdauer mancher aus organischen Materialien bestehenden Bauelemente dar. Bei (O)LEDs bezeichnet man als Lebensdauer die mittlere Betriebszeit, nach der die Leuchtdichte auf die Hälfte abgesunken ist. Für weiße Lichtquellen und Monitore ist für die insgesamt nutzbare Lebensdauer die der blauen Komponente begrenzend. 2011 wurden für weiße Lichtquellen 5000 Stunden (bei 1000 cd/m²)[16] und 12.000 Stunden (bei 100 cd/m²)[17] angegeben.

drminol, 2016-12-19:
Because it's an option. No one tells you to use this option. But certainly are people who would like to use it, even at the cost of the screen life. We shouldn't limit other people. If you don't want to damage the screen or battery simply doesn't enable this option.

sandboxgangster, 2016-12-19:
Is ok, is nicht mein Projekt war nur ne idee wenn man den Screen Saver aus hat, ist nen total weißes Display total sinlos dafür schaltet man den doch aus das gerade nix angezeigt wird.

Macht das wie ihr denkt, war ja nur ne idee

sandboxgangster, 2016-12-19:
Is ok, is not my project was only ne ideae if one has the screensaver, is nen total white display totally sinlos for it switches the nevertheless from the straight nix is displayed.

Does that as you think, was only ne idea

you can close this than

sandboxgangster, 2016-12-19:
Is ok, is nicht mein Projekt war nur ne ide wenn man den Screen Saver aus hat, ist ein weißes Display total sinnlos dafür schaltet man den doch aus das gerade nix angezeigt wird.

Macht das wie ihr denkt, war ja nur ne idee

closed 2016-12-19 by sandboxgangster