ClockSelect/myevic #214: No deep discharge protection

locomarco, 2016-12-17:
I flashed this firmware a few weeks ago on my RX200S.
Three almost fully charged Sony VTC4 were left in the device. As far as i remember, i turned it off (5-click).

When i just tried to power on the device, the screen stayed black. 
My charger shows 0.7V for one battery and 0.4V for the other two.
It seems like there is no deep discharge protection, when using this firmware.

ClockSelect, 2016-12-17:
The box always drains a few power, whatever the firmware. Turning the box off put it into an idle mode, but does not really shuts everything off. This firmware drains a little more if you have the clock software enabled (LSL option ON in the expert menu), so battery draining within weeks will go a bit faster, but not that much. Whatever the firmware, you should never store the box for weeks with batteries inside. 
Weeks timescale seems a bit fast to me; a fully charged RX200S should last 2 or 3 months at least before emptying batteries.

ClockSelect, 2016-12-17:
Anyway, a deep discharge protection cannot be done by firmware. The chipset won't boot up below 2.8V, so a deep discharge protection is really an hardware issue since the firmware as no control over it.

locomarco, 2016-12-17:
I know there will always flow a small current. 
To me it looks like the firmware doesn't put the mcu in deep sleep mode correctly. Or the LSL Mode draws a lot more current as it should.

Whatever the problem is, i haven't noticed this behaviour running the official firmware.

ClockSelect, 2016-12-18:
Based on the MCU datasheet, LSL is estimated to drain at most 50mAh/day, that's around 5 monthes for a fully charged RX200S, that's why your experience is a bit surprising.
But since I've no multimeter, I could never really check this assumption, unfortunately.

closed 2016-12-21 by ClockSelect