ClockSelect/myevic #215: Battery percentage, scroll speed

gsvincze, 2016-12-17:
The battery level is not showing correctly, the issue is that it goes linear from 4.2 volt to 3.x (haven't drained the battery fully). The lithium batteries does not have a linear discharge curve, so for example at 4.0 volt it says its at about 75% which in reality is more like 90%.

It can be made rally close to by 3 linear stages or a table, for example 4.2-3.9 is 100-90, 3.9-to 3.4 90-20, 3.4-3.0 20-0 assuming it doesn't goes below 3 volts.

Here's a basic discharge curve (ignore the lower discharge rates)
www.mpoweruk.com/images/discharge-C-rate.gif

The other issue is that the scrolling is watts or temp is over engineered and speeds up slowly, kind of anoying compared to recent recent joyetech firmwares. What I think would be much better is to have 5-10 step at slow that the others at fast.

ClockSelect, 2016-12-21:
What's your battery model ?

Joyetech firmwares speed up at an insane speed. I'm human being.

gsvincze, 2016-12-28:
Samsung 20R, pretty similar to 25R but has less capacity and higher current rating.

ClockSelect, 2017-01-12:
The discharge curves as the one you linked shows voltages under load.
The voltage you read on the box while not firing is the voltage at rest, which is different.
The box uses "charge curves" rather than "discharge curves" to display accurate state-of-charge.

I agree curves may be not perfect; I do with what I found here and there (mostly curves from DNA boxes). I don't really know how accurate they are.

I'd be happy to have a reliable source for accurate curves, but unfortunately it's not the case, and I do not have the necessary hardware to build them.