ClockSelect/myevic #218: Question about mL/kJ and emulated rtc impact on battery

USA-RedDragon, 2016-12-18:
Sorry to make an issue for a question, but I can't find a way to contact you directly

Question A: The mL/kJ value in settings, says in your doc "The proposed value of 360 is rather accurate for a standard 30W factory coil. It may need to be adjusted depending on your setup." I was wondering if you have a generic equation on how to calculate this value or something to help determine what value is needed.

Question B: I think you may have answered this before, but what is the estimated impact of rtc emulation on battery life? I've been using stock fw in fear that it will heavily impact the battery life. In fact, I thought about trying to remove it from the firmware just to avoid this, but haven't found the time, especially with all the other projects I've been doing. 

**Box Model:**
RX200S - 3x HG2s
**Coil setup:**
Using the TFV8's T10(10 coils) rated at .129 ohms and as printed on the coil it's rated at 50-300W and recommended at 130-190W

falconettt, 2016-12-18:
Answer to question A:
Just simple proportion,
**Desired value** (ml/kj) = **Set value** (ml/kj) / **Vaped as shown on the screen** (ml)} x **Really vaped**  (ml)

ClockSelect, 2016-12-21:
Yes, just that simple proportinal rule.

closed 2016-12-21 by ClockSelect