ClockSelect/myevic #219: VTC Mini - Expand 1st and 2nd line too

BigBadBlackWolf, 2016-12-19:
- Selection in the 1st line: VOLT -> VOUT -> BATT
- Selection in the second line: COIL/(COIL Lock) -> RES
- Selection in the 3rd line: AMP -> PUFF -> TIME -> BOARD (temperature) -> CLOCK (digital)
as is currently possible in the custom firmware "myevic 161107" (only without my mentioned change regarding line 1 and 2).

ClockSelect, 2016-12-21:
You already have anything you want on third line, iirc; first and second line are not yet configurable. It may come if we can make a good configuration tool; it's an interface problem. Things get rapidly complicated if we want to make everything configurable.

BigBadBlackWolf, 2016-12-21:
Okay, thanks for the quick reply. Was synonymous only such an idea / suggestion on my part with regard to line 1 and 2 synonymous configurable to make. Did not know that there is an interface problem.

ClockSelect, 2016-12-21:
Question is: how would you make everything configurable as you wish without having to click a thousand times to get what you want?
This unbearable with tree buttons, obvioulsy. Things must be kept simple.

bluenazgul, 2016-12-21:
How?
Idk how it is done in ArcticFox - but it is working there (ok, the also
have the NFE Support, that's malen it a lot more comfortable)  - maybe this
could be a solution

closed 2017-01-13 by ClockSelect