ClockSelect/myevic #220: Screen wakeup issue.

roaminghawk, 2016-12-21:
Hi,

eVic VTC Mini here V3.30 Build 161218
The screen keeps coming on when it's supposed to be off.
You click the button, the screen comes on, you vape, and you set it aside, after 10 seconds and sometimes 50 seconds (also happens randomly, sometimes it doesn't come on at all and sometimes after 5 minutes or so) the screen comes back on looking like this:
http://i.imgur.com/uRnNvWJ.jpg
Sometimes it stays on until you click the fire button and sometimes it goes away and screen lights up like you have pressed the fire button.

What can I do?

Thanks.

ClockSelect, 2016-12-21:
You mean: the screen is black, off, then after some time, without touching
anything, it shows on?

roaminghawk, 2016-12-21:
Yes.

ClockSelect, 2016-12-21:
I spend most of my life with at most one screen of a box in front of me and I never experimented this. Have you a procedure to reproduce this with some certainty?

roaminghawk, 2016-12-21:
I'm afraid not. I can't reproduce the issue. Right now it's working fine, But I will try to get on video maybe that would help.
Also when this happens the menus gets glitchy, like you click the fire and right control button and get into menus and it  pops out of the menus section without you pressing anything.
A question, What is the option to show the image I posted above?

roaminghawk, 2016-12-21:
Also, How can I reset the firmware? I mean fully erasing all the settings and cache?

ClockSelect, 2016-12-21:
The screen you've posted is the screen you normaly get when plugging on the USB cable.
The issue may come from your USB connection glitching; it's a reasonable hypothesis.
You can reset the settings with any firmware editor software, such as NFirmwareEditor. But in your case, I'm afraid it won't help.

ClockSelect, 2017-01-13:
No news, good news.

closed 2017-01-13 by ClockSelect

roaminghawk, 2017-01-13:
Yes, Sorry about that. I totally forgot about this thread.
You were correct, the USB socket was indeed defective. Once I changed it with another one, the mod is working as it should. Thanks.