ClockSelect/myevic #222: Not really an issue, where to find changelog?

quentin6633, 2016-12-22:
I searched but didn't found it, is there a changelog?
I saw you made a modification 2 hours ago, i'm using previous build on my RX2/3, working perfectly!! So i want to know if it s necessary to do this update :-)
Thanks all for the hard work it's really a good firmware, so many possibilities compared to the stock one...

bluenazgul, 2016-12-22:
It is easy, just check the commits - the Last change was "NFE Compatiblity"

quentin6633, 2016-12-22:
Thanks mate, didn't found it. Even on mobile version you can access to it?

SEEKNDESTROY, 2016-12-22:
Try this [Commits](https://github.com/ClockSelect/myevic/commits/master)

SEEKNDESTROY, 2016-12-22:
I hope this helps...
[image]

quentin6633, 2016-12-22:
Yeah mate it helps! With your link i acces to it directly, and found also where i can access to it.
On the main page, need to click on "view code" and after that here it is :-)
Thanks dudes for your help!
I'm not an expert on Github, just subscribed today :-)
I can close this "issue" now ;-)

closed 2016-12-22 by quentin6633

SEEKNDESTROY, 2016-12-22:
No problem... we were all new to this at one point.