ClockSelect/myevic #224: Compatibility for Elitar pipe?

AAI688, 2016-12-23:
**Box Model:**

is this firmware compatible with Joyetech Elitar Pipe? since it is a Joyetech product too.. however i can see the screen size is smaller so im not too sure if its okay to try or not. I would try if there is no chance of bricking it. Is there any chance of bricking?

Thanks in advance.

donicrosby, 2016-12-25:
I don't think it will be supported probably for the same reason why the original rx200 isn't, the screen is not big enough. I wouldn't try flashing this to your device because it is not on the list of supported devices it will most likely brick it.

However if someone knows if it is possible to create another version of this firmware for the smaller screens it would allow those devices to be much better.

TBXin, 2016-12-26:
Small screens is not problem, just requires some attention and time.
This CFW supports rx200, istick pico, istick100, etc.
[image]

AAI688, 2016-12-26:
is there a way to pull the current firmware from elitar also so that incase any f*ck ups it can be flash back to original?? joyetech website dont have the original firmware/flasher yet. which is one of the reasons why i havent flashed myevic/arcticfox firmware..

TBXin, 2016-12-26:
Neither myEvic nor ArcticFox are not supporting Elitar, so, even if OFW will be available you still can't upload CFW to your Elitar.

AAI688, 2016-12-26:
okay, i just wait for either to come out then.. which other new devices u said uses the elitar screen from eleaf btw? i cant seem to find any..

TBXin, 2016-12-26:
http://www.eleafworld.com/aster-rt/

closed 2017-01-11 by ClockSelect