ClockSelect/myevic #225: Preheat bug when turned off

quentin6633, 2016-12-23:
**Box Model: Reuleaux RX2/3 with vtc6 batteries

**Coil setup:Aromamizer Supreme with dual 3mm ID clapton coils

I have set the preheat at 70w, 0.5sec time and 0.20 delay.
When I vape on it, and just after I turn it off, when I turn it back on the "P" indicator is still blinking, after a long power off wait, clearly more than the delay time, i tested even with a 5 minutes powering off time.

I'm on the previous build, the one just before the NFE comp update.

The clock is set, the date also.
And i changed no setting in the expert menu

ClockSelect, 2016-12-27:
Yes it restarts the preheat delay timer where it left if at power off.

ClockSelect, 2016-12-28:
Should be fixed

quentin6633, 2016-12-28:
Nice! Thanks mate! Will try as soon as my RX2/3 is replaced, the chipset is fried, no display...

ClockSelect, 2016-12-29:
Real hardware issue?

quentin6633, 2016-12-29:
For sure tried all i can :-) you can have a look at the problem on this video https://youtu.be/f7kxq3Vm0fg

ClockSelect, 2016-12-29:
Weird!

quentin6633, 2016-12-29:
Yeah it stopped working just after firing it.
When i wake and put the batteries in, it works for like 5 minutes max, but it is very weak, no brightness, and after that only info i can display is the lock/unlock and if i don't wait for it to disapear and click on fire, the display do a kind of flash and go back to black.
Tried so many things... 4 days after receiving it's a bit frustrating....

closed 2016-12-29 by ClockSelect