ClockSelect/myevic #226: Usb charge not complete or really slow overnight

quentin6633, 2016-12-23:
**Box Model: Reuleaux RX2/3 with vtc6 batteries

**Coil setup:Aromamizer Supreme with dual 3mm ID clapton coils

I know it's not a good thing to charge trough usb, i'm waiting for my 4 slots charger so yesterday i tried the charge with usb just for one night.

I plugged it at 1 o'clock at only 10% battery left and when i waked up at 9, it was still charging, it was around 65 to 70% max, around 4 volts but i don't remind exactly.

It was plugged on a 2.4A charger, which is working perfectly.

Using the previous build, just befor the one with NFE Comp improvement

SEEKNDESTROY, 2016-12-23:
The chip has a maximum charge input of 1 amp to protect the user and the chip from overheating and possibly catching on fire. In the past, people used 2+amp power bricks/chargers on unprotected devices...causing some of the devices to vent or "explode".

quentin6633, 2016-12-23:
Good to know!
But even at 1A it shouldn't be charged if it's plugged in for 8 hours?
It seemed blocked when i waked up, still the charging indicator but no movement on the voltage or percentage

SEEKNDESTROY, 2016-12-23:
3 vtc6 batteries = 9,000 mAh or 9 Amps. Lets assume your are actually charging at 1 amp...depending on the age of the cells, it would take anywhere from 9 to 12 hours to fully charge the batteries.

quentin6633, 2016-12-23:
I see! I didn't thought this way.. So you're true!! it's longer than i thought...
They are new, bought two days ago :-)

SEEKNDESTROY, 2016-12-23:
I like and use the [efest luc v4](https://www.amazon.com/s/ref=nb_sb_noss?url=node%3D667846011&field-keywords=efest+luc+v4), you can charge 4 batteries at 0.5 and 1 amp and 2 batteries at 2 amps.

quentin6633, 2016-12-23:
I'm receiving tomorrow my Xtar VC4 same kind of charger :-)
I have a Nitecore i2 so 2 slots only so it requires some time to do all batteries... ;-)

closed 2016-12-27 by ClockSelect