ClockSelect/myevic #227: evic-usb cannot find device

vilu85, 2016-12-25:
I can't find way to get evic-usb working. Everything compiles fine but looks like Nuvoton driver doesn't work. It says that the device cannot be started (error 10).

Configuration in use:

OS: Windows 10 (x64)
Terminal: Cygwin
Inf: [https://github.com/ClockSelect/myevic/blob/master/driver/NuvotonCDC.inf](url) USB\VID_0416&PID_5020&MI_01
> #### Setup details from Windows Event Log:
>   DriverName nuvotoncdc.inf_amd64_34898a1a3e74ad2c\nuvotoncdc.inf
>   DriverVersion 1.0.0.1  
>   DriverProvider Nuvoton Co., Ltd. 
>   DeviceInstanceID ROOT\UNKNOWN\0000 
>   SetupClass {4D36E978-E325-11CE-BFC1-08002BE10318} 
 
[image]
Device is on the list however, *Bus 002 Device 003: ID 0416:5020 Winbond Electronics Corp.*

**Box Model:** eVic VTC Dual

vilu85, 2016-12-25:
[image]

dywersant, 2016-12-25:
Enable HID in expert options.

vilu85, 2016-12-25:
Where are expert options? I just cannot find them :O

SEEKNDESTROY, 2016-12-25:
Press **Power** and **Plus** > Press **Minus** Twice > Press **Power** > Press **Power** until **USB** displays **HID** > Press **Minus** one time Press **Power** to exit

vilu85, 2016-12-25:
That didn't solve the issue either. I think the problem is in Nuvoton CDC driver, it doesn't communicate with actual USB port and evic device. Other people have also problems with VCOM drivers in Windows 8.1 to Windows 10. That's what I found by Google.

Can someone confirm that Nuvoton driver still works in Windows 10 (x64) and is it normal that the driver shows "Device cannot be started (Error 10)" in Device Manager -> Nuvoton Virtual Com Port (COM3) -> Properties?

SEEKNDESTROY, 2016-12-25:
I have Microsoft Windows 10 Pro - Version 10.0.10586 Build 10586
I don't have nuvoton virtual com port (COM3) installed... I have "Device Manager - Intel(R) USB 3.0 eXtensible Host Controller - 1.0 (Microsoft)"

Try reinstalling the driver or try a newer version like "COM7". Google [nuvoton virtual com port (COM3) windows 10 driver download](https://www.google.com/search?q=Nuvoton+driver&ie=utf-8&oe=utf-8#q=nuvoton+virtual+com+port+(COM3)+windows+10+driver+download)

vilu85, 2016-12-25:
I thought Nuvoton VCOM driver must be installed if you want to use python-evic?

vilu85, 2016-12-25:
I found the issue! First I removed all hidapi and python-evic related stuff from Cygwin, then reinstalled everything and that's it. Now it works!
So the issue was maybe some error happened earlier in python-evic setup or hidapi patching/setup.
Closes #227

closed 2016-12-25 by vilu85