ClockSelect/myevic #229: Linux: autosync on device connected

Nefelim4ag, 2016-12-25:
As Issues on python-evic are closed, i want to place info to this issue %)
I packaged python-evic for aur:
https://aur.archlinux.org/packages/python-evic-git/

For autosync on connect:
```
    {
        echo '[Unit]'
        echo 'Description=Evic RTC sync'
        echo '[Service]'
        echo 'ExecStart=/usr/bin/evic-usb time'
        echo 'RemainAfterExit=yes'
    } > "$pkgdir/usr/lib/systemd/system/evic-usb-rtc-sync.service"

    {
        echo "# HIDAPI/libusb RTC Sync"
        echo 'ACTION=="add", SUBSYSTEM=="usb", ATTRS{idVendor}=="0416", ATTRS{idProduct}=="5020", RUN+="/usr/bin/systemctl restart evic-usb-rtc-sync"'
    } > "$pkgdir/usr/lib/udev/rules.d/99-nuvoton-hid-rtc-sync.rules"
```
AFAIK external service for sync time is needed only for systemd systems, as systemd add some restrictions to udev and evic-usb can't see device if running as script in udev.

JFYI, you can close this or say me for do a pull request. Thanks.

ClockSelect, 2016-12-29:
Did you try to post the issue on the Ban3/python-evic repo? Mine is just a fork.
Autosync is a nice feature, indeed.

Nefelim4ag, 2016-12-29:
@ClockSelect, oh, thanks

closed 2016-12-29 by Nefelim4ag