ClockSelect/myevic #230: Compatibility for RX Mini, please!

krudas, 2016-12-27:
**Box Model:**

**Coil setup:**

ClockSelect, 2016-12-28:
Should work, but could not test.
Returns welcome :)

krudas, 2016-12-29:
Sorry m8, but doesn't seem to work, still error

ClockSelect, 2016-12-29:
What kind of error?

ClockSelect, 2016-12-29:
Try this one... never knows.
Since the box ID appears several times in the firmware file, the uploader may be mistaken.
This could be the issue.

krudas, 2016-12-29:
[image]
Still "The firmware file is error"

krudas, 2016-12-29:
Ok, now I downloaded the master file and extracted the .bin from that. Now it works so far, but the clock doesn't accept any setups. I suggest the RX Mini doesn't have any RTC onboard ...

ClockSelect, 2016-12-29:
Yes, RXmini has no built-in RTC so using RTC features may have random results.

commit 046e8859 (2016-12-29): [#232] Fix fix #230
closed 2016-12-29 by ClockSelect