ClockSelect/myevic #234: RX2/3 'Please confirm hardware version'

Shure123, 2016-12-30:
Hi,

I cannot update my new RX2/3 with the myevic.bin firmware. I keep getting this **'please confirm the hardware version!!'** message:  http://imgur.com/3TUfYNO

_Hardware version is 1.02_. 

- I can succesfully upgrade and downgrade between 4.03 and 4.13, so the updater and usb cable work.

- I can also upgrade my eGrip2 with the same bin, so the file is not corrupt.

- I cannot find any information about this 'please confirm the hardware version' message, nor do I know how to confirm it. It only gives an 'ok' button which just returns me to the begin screen. 

- I tried with the device on & off, by connecting before & after opening the updater, and with & without batteries installed. 

**Box Model:** RX2/3, hw version: 1.02, fw version: 4.13 (+ 4.03)

[image]

Thank you for reading.

ClockSelect, 2016-12-30:
myevic is not compatible with RX2/3 hardware version 1.02 (yet) since this hardware did not exist at the time RX2/3 compatibility has been developed.
So I don't recommend you flash your box with myevic untill I check for RX2/3 compatibility.

ClockSelect, 2016-12-30:
There something strange, there: if I take a look at official Wismec firmwares, they're supposed to handle only up to 1.01 hardwares. So you shouldn't even be able to downgrade to 4.13.
When did you got your box?

ClockSelect, 2016-12-30:
I had a look and Wismec silently changed their firmware files to include hardware 1.02 without changing firmware versions and dates. Neat.

Shure123, 2016-12-30:
Thanks for your fast answer. I am really liking myevic (big thanks!). So far everything works fine, I'll test it some more and can report any issues.

If you like, I can remove the workaround from my post until you deem it safe though.

ClockSelect, 2016-12-30:
It seems that there's no significant changes in firmware related to hardware version.
At least nothing dangerous.

easyyu, 2016-12-31:
I have same error message..
After I try to install firmware my VTC VTwo is dead.
Only with firmware before this I can flash it and make it work again.
What can I do to make work with latest firmware?
Is it a bug?
[image]

Shure123, 2016-12-31:
@easyyu Try to press all 3 buttons after flashing when it is 'dead'. For me that brought it back to life.

@ClockSelect I ran into the first issue. It's minor, but considering your name, it might bother you more than others ... Clock: SetTime and SetDate don't work. I can 'change' the time and date, but it doesn't change anything after I exit the menu. The date is still 2084. The time can be changed with ClckAdjust but that's tiresome.

SEEKNDESTROY, 2016-12-31:
@Shure123 long press the power button to save the date.

quentin6633, 2017-01-01:
Mine says also hardware version 1.02, but was able to install the firmware without problem.
Can my problem of fried chipset or chipset come from here? After this problem, tried to install all the official firmwares available so it's a not a soft problem for sure.
I'm receiving the new one tomorrow will wait before installing :-)

Shure123, 2017-01-01:
@JOOKERO 1.00 is the hardware version of your RX2/3. That won't ever change. 
To check the software version, click the fire button 20 times.

JOOKERO, 2017-01-01:
it shows : (myevec , build 161230 , version 3.30) , with NFirmwareditor i have 1.02 hardware version , would you tel me the fiffrence between 1.00 and 1.02 version ??

easyyu, 2017-01-08:
I want to ask development, why I can not use from aprox 22.Dec.2016 release any more this firmware on my Evic VTwo mini. 
I always get error: Please confirm the hw version.
My Vtwo have hw verision v1.01, readed by FWupdater v1.2
Thx

ClockSelect, 2017-01-11:
@easyyu VTwo Mini 1.01 is a new hardware version.
I have to check for its compatibility.

ClockSelect, 2017-01-11:
@JOOKERO : I've seen no firmware-related difference between the two; maybe they just changed some random component on the PCB.

easyyu, 2017-01-11:
> >  ClockSelect commented 7 hours ago
> 
> @easyyu VTwo Mini 1.01 is a new hardware version.
> I have to check for its compatibility.
> 
Thx for answer.
But then why I can not use anymore fw you made?

ClockSelect, 2017-01-12:
Now :)

easyyu, 2017-01-12:
> @ClockSelect
> 
> It's because HW version was badly checked before.
> You should be able to use it again, now :)

Thx mate..now I can again use your FW!

closed 2017-01-18 by ClockSelect