ClockSelect/myevic #235: are you adding the rx300 atall as love your mod

docdeath, 2016-12-30:
**Box Model:**

**Coil setup:**

ClockSelect, 2016-12-30:
I didn't look at the RX300 yet; since it's an unusual multi-cell box, it may (certainly will) take some time to adapt.

ClockSelect, 2017-01-12:
For those of you ready to take some risks, I made [an experimental RX300 version of the firmware](https://www.dropbox.com/s/6878hw1w0ufei1t/myevic.bin?dl=1).
Charging via USB is not yet enabled; it's the most dangerous feature and is not ready yet.

Since I have no RX300 by hand I cannot test by myself so I must rely on the good wills down there.

docdeath, 2017-01-12:
it loads fine and works as in you can use menus but it comes up with flashing battery icon  at bottom and when you hit fire button it say check  battery's i changed to fully charged set dose same .menu seams to all work  apart from  it will not fire

ClockSelect, 2017-01-12:
What do the battery voltages of read for each of the four cells?
You can see them in Expert->BVO menu.

ClockSelect, 2017-01-12:
Ok I found a bug in battery reading.
You can [retry](https://www.dropbox.com/s/6878hw1w0ufei1t/myevic.bin?dl=1).

ClockSelect, 2017-01-12:
Thank you for your help, BTW^^

docdeath, 2017-01-12:
that has solved  battery problem now as you hit fire it says device to hot  flashing up

docdeath, 2017-01-12:
no problem with helping as love your firmware

docdeath, 2017-01-12:
i have been though all the menus now and as far as i can tell it all works  .just the device to hot when you hit fire button

ClockSelect, 2017-01-13:
What's the board temp?

docdeath, 2017-01-13:
185 c then i hit fire button to see broad  temp 10  times and its now reset self to 30 c

ClockSelect, 2017-01-14:
I found a bug in temp sensing. It should work better.

docdeath, 2017-01-14:
wheres bin file  ? to test

ClockSelect, 2017-01-14:
[Here](https://www.dropbox.com/s/6878hw1w0ufei1t/myevic.bin?dl=0), that's always the same link.

docdeath, 2017-01-14:
ok my bad thanks well temp problem solved now and working ok so far  only thing i cant seam to get  10 clicks for temp up 20 clicks for firmware works but 10 clicks nothing happens will keep trying as its always been hit and miss with that one as far as i can tell it all works clock works  and every thing else in menus works

ClockSelect, 2017-01-14:
Fine, I released it. We can wait a bit for USB charging - not an essential feature on a 4 cells box, anyway.

docdeath, 2017-01-14:
well charging it with usb is not good thing anyways  best to remove and charge in proper charger , have tested with sock firmware using usb to charge and it very in balanced so  would not use

ClockSelect, 2017-01-18:
[New prototype version](https://www.dropbox.com/s/6878hw1w0ufei1t/myevic.bin?dl=1), with supposedly functional battery charging, including balancing.

Note about balancing: the box balances the load at *end of charge*, that is when one of the cells has been fully charged and others are still late. Half-charging will not balance. Balancing should set cells within 0.03V from each others.

Testers needed.

ClockSelect, 2017-01-19:
@docdeath Did you test this one?

docdeath, 2017-01-19:
@ClockSelect  AM testing at moment  will let you know asap

docdeath, 2017-01-19:
@ClockSelect  it all seems to be working as it should  not fully charged  as yet but  battery's are  close to each other tested with dig volt meter .04 will give full charge tomm and see if any changes

docdeath, 2017-01-20:
@ClockSelect fully charged 30q Samsung with all battery's with in 0.5 off each other

ClockSelect, 2017-01-21:
I hope you meant 0.05^^
Thank you!

ReMicroN, 2017-01-21:
Does it works with RX2/3 or RX200S?

ClockSelect, 2017-01-21:
RX2/3 and RX200S work as usual. Cell balancing is specific to the RX300.

ReMicroN, 2017-01-21:
So RX300 has balancing IC?

ClockSelect, 2017-01-21:
Sort of

closed 2017-01-31 by ClockSelect