ClockSelect/myevic #237: Suggestion: rx300 Reverse Charging on all box

ExperienceTesla, 2017-01-02:
with the rx300 wismec har releaseda new feature: reverse charge. if you put this adapter in 510 connector of the box, connect your phone with the usb cable and you press fire, the box star charge the phone
http://www.wismec.com/accessories/avatar-reverse-charging-adapter/
if you can and if you want you can put this feature on your custom firmware.
thanks for the attention
p.s. sorry for my bad english

ReMicroN, 2017-01-03:
First we need to disassemble this charging adapter to see what element is used to make boxmod see it as adapter rather than the atomizer. This may be a specific high resistance or something other. Then we need to write a subprogram which will provide a stable 5V voltage output to 510 pin...

ClockSelect, 2017-01-04:
I'd love pictures.

coolfocks, 2017-01-04:
Great feature, though I can't wait until we start using our mods as phones as well

enricorov, 2017-01-04:
Nothing fancy, you set the regulation to 5V, then connect the phone USB to
the 510 connector.

docdeath, 2017-01-04:
i have rx300 but just hoping some time soon your update your custom firmware to use on rx300 as there firmware feels like stone age stuff lol

ClockSelect, 2017-01-12:
Multi-cell boxes are always tricky to adapt on.
For the reverse charger, seriously, I'd prefer my phone to be capable of charging my e-cig.

Justin-L-Franks, 2017-01-13:
"For the reverse charger, seriously, I'd prefer my phone to be capable of charging my e-cig." Agreed. I'd much rather trust my ~$500 phone to safely charge my ~$50 box mod then vice-versa.

MrScaryZ, 2017-01-30:
I think this is a waste of time really nobody uses this function this is not the first time mods have had this capability its just a gimmick and if it works great but ask yourself a question if you can get a dedicated usb charging box for 15,00 from china why would anyone want to use your 'MOD' that you are vaping with to charge your phones its just silly really

ExperienceTesla, 2017-01-30:
> @ClockSelect you can charge your box with your phone with otg adapter (not supported on all phones)
@MrScaryZ i think this feature is only for emergency

enricorov, 2017-01-30:
It is *not* thought as an emergency feature  but more like an useless feature to set themselves apart from competitors. 

Even taking the matter seriously, having to carry a 510/USB adapter negates the emergency use case scenario. 

As for reliability, though, it's just a switching regulator, and USB protocol allows 10% voltage fluctuation, and since a phone would be drawing a constant current, I don't foresee any problem in the feasibility of such a feature.