ClockSelect/myevic #238: Issue reading voltage.

Clavisnl, 2017-01-05:
Hi!
2 days ago i recieved the mod from fasttech. All good. Today the battery's ran out, so i replaced them. Hmm, also empty? WTF? So i put them in my other mod, and they were full. Wich i knew.

So i guess there's something badly wrong with my reuleaux.. I opened the mod, but there are no loose soldering connections or whatsoever.

Tried to reflash older and newer firmware, but no results.

H/w version 1.02

I tried with the myevic software, but still the same problem. Even after BVO, wich only can handle +/- 30mv. My batteries are like 3.3 and 4.7, althrough i use them in my Alien aswell, it works fine.
Mod is also not charging by usb cable, i can update firmware though.

So not a problem with your software, wich runs great. But ill hope you or anyone can help me.

Shure123, 2017-01-05:
I have tried to assist @Clavisnl  with this problem on a Dutch forum, but also suggested him to open an issue here on GitHub.

I have the same the problem that my RX2/3 with hardware 1.02 detects the voltages incorrectly. For me, the difference is 0.7-0.8V. Both batteries measure exactly the same on both my multimeter and analytic charger. But the RX detects the top one as 0.7 more. Even when I switch places, it still detects the first one als having 0.7V more. Which is clearly impossible.

So for me the difference is solved bij changing the offset, but for Clavis the difference is too large for the +/-300 limit. Maybe he could edit the line of code that limits the offset? I understand that ClockSelect wants to keep the limit in place for security, but the RX2/3 with hw 1.02 measures very incorrectly, so the offset limit might be too strict in this particular case.

Now neither Clavis nor I are comfortable changing the code without any guidance, so if possible, a small explanation on how he could change it so he can bring them closer together would be welcome.

The numbers: before installing myevic and changing the battery profile & offset, batt. 1 gave 3.34, batt. 2 4.71, with a 15% battery level.

After changing batt. profile and offset, batt. 1 gives 3.64, the other 4.41, with a battery level of 45%. Both with full batteries testing at 4.2V on a meter, and giving 100% on a Smok Alien.

The difference went down from around 1.4 to around 0.8, but the gap is still too large for the RX to work correctly.

JOOKERO, 2017-01-05:
you can read exact voltage by other way : make the box off , then hold fire+right button for 3 seonds . you can also try with 2 batt. the cover may cause problems .

Clavisnl, 2017-01-05:
Yea i know, thats what i did. I only use 2 batteries. With 3 batteries it gives me the same crap...

JOOKERO, 2017-01-05:
myevic is the most complet firm. but with VTCFont i found a modified firm for RX2/3 and when in charge hit fire 3 time and it will affich votage in and out and all batt volt , finaly you aybe have to change the USB cable , i tryed that and it works .my box was fine and the usb cable was the probleme

ClockSelect, 2017-01-12:
RX2/3 are known to be inaccurate at reading things, but it's hard to imagine it could get worse with a newer hardware version.

If the box reads voltage inaccurately, I wouldn't recommend to charge it via USB since it may overcharge the cells (USB charging heats the box and also takes ages, so not a good idea in the first place). Disable USB charging in the expert menu (UCH OFF) and use an external charger.

Maybe I could extend the BVO range, but it's a rather ugly way to compensate for a bad hardware.

Justin-L-Franks, 2017-01-13:
I had the same thing happen to me once with my Cuboid. It was working fine, then suddenly both batteries showed as empty. Freshly-charged batteries also showed as empty. Turns out the battery contacts were dirty -- they looked clean, but there must have been some residue on them from somewhere, because cleaning the contacts fixed it. The box was running the stock firmware at the time.

This might not help, but it's worth a shot.

ClockSelect, 2017-01-14:
I've upped the max BVO, in case it helps.

Concerning RX2/3 battery reading issues, I'd recommend to check the internal wire soldering; there is a flat copper wire connecting the top and the bottom of the box that causes many issues. You can find many videos for this on youtube.

closed 2017-01-18 by ClockSelect