ClockSelect/myevic #239: Anyway to change puff counter, coming from stock software

danielthelad, 2017-01-05:
Hey, I want to update to myvevic software on my RX200S but I don't want my puff counter reset, I'm currently around the 30k mark but updating to this software will reset it, is there anyway to change it? Also if it is possible it wouldn't be stuck at 30k would it?

**Box Model:**RX200S

**Coil setup:**TFV8 T10

ClockSelect, 2017-01-12:
Normaly, the puff counter does not reset on software update.

danielthelad, 2017-01-12:
It does reset, it happens every time you update it unless it's the same file version in which case this one is not and it would go back to 0. Noticed another thread where someone put in a submenu where he could change them manually

ClockSelect, 2017-01-12:
It should not.
It doesn't reset for me, and I update my boxes constantly with new versions.

danielthelad, 2017-01-12:
I've just done it and it reset it to 0 puffs, is there no way it can be changed?

ClockSelect, 2017-01-13:
What are you doing exactly?
Are you just upgrading from a myevic version to another?

danielthelad, 2017-01-13:
Just upgrading from stock wismec firmware to myevic software, thats why its been reset as it has different version number

danielthelad, 2017-01-13:
Hey it don't matter now anyway as I have sold the device but I do have an RX300 now, will that be supported soon and will the contrast be able to be increased as the screen is so dark on Wismec stock software

ClockSelect, 2017-01-14:
You may have a look to #235.

enricorov, 2017-01-14:
Just to comment on the issue, I just updated from a November build to the latest and the puff counter was not reset, so it may have been an issue on @danielthelad 's side.

danielthelad, 2017-01-14:
I mean I was coming off wismec software to my evic software so it was reset then. I was figuring out if anyone could try and get my puff counter on stock software to this custom firmware

ClockSelect, 2017-01-14:
There's currently no way; but there's a configuration editor coming soon that should be able to do it.

danielthelad, 2017-01-14:
Oh that's good, thankyou. Will look forward to myevic when it's fully supported for the RX300, will I able to edit the contrast on the RX300 because it has a very dark screen

gpolydoros, 2017-01-18:
I think this has to do with the application performing the update. I use NFE to update and indeed the puff counter (and any other setting for that matter) does not reset. Most settings like wattage and temperature (most likely all *applicable* settings) are retained even when going from stock firmware to myevic and then back to stock again. Maybe it's WISMEC's software that triggers a settings reset after updating the firmware.

@danielthelad Careful when changing the contrast of your mod. If you use anything over 17% with stealth mode disabled, certain pixels will start appearing darker after a while. In fact, even 17% is a lot for pixels that are constantly used (like the battery status when charging). I now have it set to 5%, which is fine for indoor use.

ClockSelect, 2017-01-26:
@danielthelad RX300 is fully supported now.

danielthelad, 2017-01-27:
@ClockSelect that's brilliant thank you, I will give it a try soon